mavlink/qgroundcontrol #4801: Spontaneous crash in Map tile code

DonLakeFlyer, 2017-03-19:
I wasn't doing anything in QGC. I think I may have done to Offline Maps.

```
//-----------------------------------------------------------------------------
void
QGeoTiledMapReplyQGC::timeout()
{
    if(_reply) {
        _reply->abort();
    }
}
```

Crashed on ```_reply->abort();```. _reply probably null but could get it to show up in the debugger.

DonLakeFlyer, 2017-03-19:
I believe it is caused by offline maps. Most likely some sort of bad shutdown sequence through the controller of something? You move away from offline maps and things are shutdown right?

DonLakeFlyer, 2017-05-08:
@dogmaphobic Did you fix this already?

dogmaphobic, 2017-05-08:
Yes, it went in in f0007592761fabab278577ef6a498f2d12d72a5a

closed 2017-05-08 by dogmaphobic