mavlink/qgroundcontrol #4803: Linux AppImage builds don't seem to work with gstreamer

DonLakeFlyer, 2017-03-19:
Tested on both 14 and 16 Ubuntu. Both spit this out
```
VideoReceiver::start() failed. Error with gst_element_factory_make('h264parse')
VideoReceiver::start() failed
Map Cache in: "/home/parallels/.cache/QGCMapCache300" / "qgcMapCache.db"
Orphaned log file count 0
LinkManager mavlinkStatus:channel:flags 0xfc0ac8 1 2
VideoReceiver::start() failed. Error with gst_element_factory_make('h264parse')
VideoReceiver::start() failed
VideoReceiver::start() failed. Error with gst_element_factory_make('h264parse')
VideoReceiver::start() failed
```

DonLakeFlyer, 2017-06-27:
@jaxxzer Says this is working fine now

closed 2017-06-27 by DonLakeFlyer

dharma-kc, 2017-10-04:
I am still getting the same error on Ubuntu 16.04. Any suggestion will be appreciated.

jaxxzer, 2017-10-04:
@dharma-kc Can you try:

`apt install gstreamer1.0-plugins-bad gstreamer1.0-libav`

and let us know if it works after installing these two packages?

See also [here](https://github.com/mavlink/qgroundcontrol/issues/4303#issuecomment-315445212).

dharma-kc, 2017-10-05:
Great. It works. Thank you.

petmonkey, 2018-05-17:
Worked for me too! Thx!!!

PavanMadushanka, 2019-02-16:
This solved it! Thanks..