mavlink/qgroundcontrol #4822: Parameter load doesn't work over UDP

lovettchris, 2017-03-23:
UDP does not guarantee delivery, or order of delivery, and so QGroundControl usually cannot load parameters from my drone over UDP which is very inconvenient.  Turns out the mavlink param_value messages do contain index and count, and it is possible to request a single missing parameter by index, so what if you changed QGroundControl to find out what is missing and request those individually to fill in the blanks.  This is what I do in my AirSim mavlink library and it seems to work ok.  Some parameters just don't exist like 13 and 269 and 525, but that's ok, it will just timeout on those and continue on.

Soimething like this:

```
std::vector<MavLinkParameter> MavLinkNodeImpl::getParamList()
{
	std::vector<MavLinkParameter> result;
	bool done = false;
	Semaphore paramReceived;
	bool waiting = false;
	int paramCount = 0;
	int paramIndex = 0;

	auto con = ensureConnection();
	int subscription = con->subscribe([&](std::shared_ptr<MavLinkConnection> connection, const MavLinkMessage& message) {
		if (message.msgid == MavLinkParamValue::kMessageId)
		{
			MavLinkParamValue param;
			param.decode(message);
			MavLinkParameter  p;
			p.index = param.param_index;
			p.type = param.param_type;
			char buf[17];
			std::memset(buf, 0, 17);
			std::memcpy(buf, param.param_id, 16);
			p.name = buf;
			p.value = param.param_value;
			result.push_back(p);
			paramIndex  = param.param_index;
			paramCount = param.param_count;
			if (param.param_index == param.param_count - 1)
			{
				done = true;
			}
			if (waiting) {
				paramReceived.post();
			}
		}
	});

	//MAVLINK_MSG_ID_PARAM_REQUEST_LIST	
	MavLinkParamRequestList cmd;
	cmd.target_system = getTargetSystemId();
	cmd.target_component = getTargetComponentId();
	sendMessage(cmd);

	while (!done) {
		waiting = true;
		if (!paramReceived.timed_wait(3000))
		{
			// timeout, so we'll drop through to the code below which will try and fix this...
			done = true;
		}
		waiting = false;
	}
	con->unsubscribe(subscription);

	// note that UDP does not guarantee delivery of messages, so we have to also check if some parameters are missing and get them individually.
	std::vector<size_t> missing;

	for (size_t i = 0; i < paramCount; i++)
	{		
		// nested loop is inefficient, but it is needed because UDP also doesn't guarantee in-order delivery
		bool found = false;
		for (auto iter = result.begin(), end = result.end(); iter != end; iter++)
		{
			MavLinkParameter p = *iter;
			if (p.index == i) {
				found = true;
				break;
			}
		}
		if (!found) {
			missing.push_back(i);
		}
	}

	// ok, now fetch the missing parameters.
	for (auto iter = missing.begin(), end = missing.end(); iter != end; iter++)
	{
		size_t index = *iter;
		MavLinkParameter r;
		if (getParameterByIndex(static_cast<int16_t>(*iter)).wait(2000, &r)) {
			result.push_back(r);
		}
		else {
			Utils::logMessage("Paremter %d does not seem to exist", index);
		}
	}

	std::sort(result.begin(), result.end(), [&](const MavLinkParameter & p1, const MavLinkParameter & p2) {
		return p1.name.compare(p2.name) < 0;
	});

	this->parameters_ = result;

	return result;
}

```

DonLakeFlyer, 2017-03-23:
The QGC parameter protocol is hardened against lossy comm links. If you want to see exactly what is going on. Look here for logging options: https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/Support/Support.html. Turn on ParameterManagerLog and look at the console output. If you aren't using a daily build, switch to that since much work has gone into param protocol hardening since last stable.

lovettchris, 2017-03-23:
Oh, cool, maybe I just need a new version then.  Looks like it is complaining about the same parameters that I found were missing.  Is it possible my PX4 just doesn't define these, and could that trip up QGroundControl?

[image]

DonLakeFlyer, 2017-03-23:
You need to run latest daily build.

lovettchris, 2017-03-23:
Same problem with latest daily build:
[image]

```
[D] at ..\src\QGCApplication.cc:269 - "Settings location "C:/Users/lovet/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at ..\src\QGCLoggingCategory.cc:98 - "Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n""
[E] at :0 - "serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT."
[D] at ..\src\Vehicle\MAVLinkLogManager.cc:342 - "MAVLink logs directory: "C:/Users/lovet/AppData/Roaming/QGroundControl.org/QGroundControl/MAVLinkLogs""
[D] at ..\src\QtLocationPlugin\QGCMapEngine.cpp:205 - "Map Cache in: "C:/Users/lovet/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[E] at qrc:///qml/QGroundControl/Controls/GeoFenceMapVisuals.qml:149 - "qrc:///qml/QGroundControl/Controls/GeoFenceMapVisuals.qml:149:29: Unable to assign [undefined] to QGeoCoordinate"
[D] at ..\src\comm\MAVLinkProtocol.cc:450 - "Orphaned log file count 0"
[D] at ..\src\comm\LinkManager.cc:184 - "LinkManager mavlinkStatus:channel:flags 0x11462ec 1 2"
[D] at ..\src\comm\MAVLinkProtocol.cc:420 - "Temp log "C:/Users/lovet/AppData/Local/Temp/FlightData756201.mavlink""
[E] at qrc:///qml/QGroundControl/Controls/GeoFenceMapVisuals.qml:149 - "qrc:///qml/QGroundControl/Controls/GeoFenceMapVisuals.qml:149:29: Unable to assign [undefined] to QGeoCoordinate"
[D] at ..\src\comm\LinkManager.cc:184 - "LinkManager mavlinkStatus:channel:flags 0x1146308 2 2"
[E] at ..\src\FactSystem\ParameterManager.cc:1048 - ""V:1" The following parameter indices could not be loaded after the maximum number of retries:  "13, 269, 525""
```

DonLakeFlyer, 2017-03-23:
> Look here for logging options: https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/Support/Support.html. Turn on ParameterManagerLog and look at the console output. If you aren't [...]

I need you to do this.

lovettchris, 2017-03-23:
log is attached.

[qgc-param.zip](https://github.com/mavlink/qgroundcontrol/files/864169/qgc-param.zip)

lovettchris, 2017-03-23:
Ok, never mind, I did a full clean rebuild of PX4 firmware and this problem is gone.  It appears that PX4 can't do incremental compilation reliably...

closed 2017-03-23 by lovettchris