mavlink/qgroundcontrol #4824: joystick button action for gear

Stifael, 2017-03-23:
It would be nice to have a button configuration for gear lif/down when using joystick

dagar, 2017-03-23:
I don't believe PX4 currently has a way to deploy landing gear over mavlink.
MAV_CMD_AIRFRAME_CONFIGURATION sounds appropriate.

Then on the QGC side we should expand the set of predefined commands the are available for joystick.
Things like gear, camera control, etc.

tubeme, 2017-03-28:
I wanted to use retracts but there is no current way to do it except using the AUX controls in the mixer and then automate it with set servo...

It would be nice to have landing retracts option.

The other problem I see is there is no good implementation for RC Joystick control exchange. In the current implementation there is no way you can use RC and Joystick in parallel, which means you have to only use Joystick if you want your Joystick buttons to work properly. Otherwise the RC is having the control all the time.

Because we are using a pro ground station we can test a lot the Joystick operations in parallel to RC. There is definitely a room for improvement in this area. Our idea was to have a software toggle switch in QGC that can be used to transfer the control between the two.

standmit, 2021-11-08:
> I don't believe PX4 currently has a way to control landing gear over mavlink. MAV_CMD_AIRFRAME_CONFIGURATION sounds appropriate.
> 
> Then on the QGC side we should expand the set of predefined commands the are available for joystick. Things like gear, camera control, etc.

QGC is not only for PX4. ArduPilot currently supports MAV_CMD_AIRFRAME_CONFIGURATION

standmit, 2022-06-23:
@dagar , could you check PR #10325 ?

closed 2026-10-05 by github-actions[bot]