mavlink/qgroundcontrol #4825: Recorded video is choppy using MKV output

anton-rothwell, 2017-03-23:
To improve recorded video output, I've had success with the following GStreamer pipeline:

gst-launch-1.0 -e udpsrc port=1234 caps="application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264" ! rtph264depay ! avdec_h264 ! tee name=tp tp. ! queue ! autovideosink sync=false tp. ! queue ! x264enc speed-preset=3 tune=zerolatency bitrate=3000 key-int-max=0 ! mp4mux ! filesink location="d:/temp/mystream2.mp4" sync = false

You will need to get rid of the h264parse in your existing pipeline as it doesn't seem to work with the mp4mux.

sync = false is important for getting smooth low latency video.

Note: avdec_h264 is in the ugly plugins (for 1.52) but works just fine.

jaxxzer, 2017-04-18:
I don't think this has anything to do with the matroska format. Are you sure that the recorded video is choppy, and that it's not just the live stream? Can you post an example of what it looks like when using master.

NaterGator, 2017-04-18:
@anton-rothwell  you'd be much better off teeing the depayloaded H264 data directly from the RTP feed and passing that directly to a muxer. You're introducing an enormous performance (and slight image quality) penalty to decode the H264 stream only to immediately reencode it. 

I have found MPEG2 transport streams to be preferable, but any robust container mux element should work well.

anton-rothwell, 2017-04-20:
@NaterGator 
Yes you're correct.  Changing the pipeline to this:

**gst-launch-1.0 -e udpsrc port=1234 caps="application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264" ! rtph264depay !  tee name=tp tp. ! queue ! avdec_h264 ! autovideosink   fps-update-interval=1000 sync=false tp. ! queue ! h264parse ! mp4mux name=muxer ! queue !filesink location="d:/temp/mystream.mp4" sync=false**

also produces good results.

@jaxxzer 
Yes the recorded video was poor and choppy.  Live stream was fine.

closed 2026-10-05 by github-actions[bot]