mavlink/qgroundcontrol #4840: Error compiling Android QGC on Ubuntu (x64)

chinanewer, 2017-03-24:
Warning: Android platform 'android--1' does not exist in SDK.
Generating Android Package
  Input file: /home/newer/build-untitled6-Android_for_armeabi_v7a_GCC_4_9_Qt_5_7_1-Debug/android-libuntitled6.so-deployment-settings.json
  Output directory: /home/newer/build-untitled6-Android_for_armeabi_v7a_GCC_4_9_Qt_5_7_1-Debug/android-build/
  Application binary: /home/newer/build-untitled6-Android_for_armeabi_v7a_GCC_4_9_Qt_5_7_1-Debug/libuntitled6.so
  Android build platform: android--1
  Install to device: No
Can't open file: /home/newer/build-~~~-Android_for_armeabi_v7a_GCC_4_9_Qt_5_7_1-Debug/android-build/gradle/wrapper/gradle-wrapper.properties for writing
11:40:18:  "/home/newer/Qt5.7.1-android/5.7/android_armv7/bin/androiddeployqt"退出，退出代码 14 。
Error while building/deploying project untitled6 (kit: Android for armeabi-v7a (GCC 4.9, Qt 5.7.1))
When executing step "Build Android APK"

in “bulid setting”page，-->"Build Android Apk"--->"Android Build SDK", can't find anyone

DonLakeFlyer, 2017-03-24:
Something is wrong in your setup. If you look at .travis.yml you can find how the build CI system builds the android build. Replicate as much of that as you can and you should have something that works.

closed 2017-03-24 by DonLakeFlyer

zhaolianggoods, 2019-05-20:
Problems with link：
obj\moc_QGeoTileFtcherQGC.obj: No such file or directory

zhaolianggoods, 2019-05-20:
Problems with Android Link:
error: obj\moc_QGeoTileFtcherQGC.obj: No such file or directory

DonLakeFlyer, 2019-06-04:
Build problems which are specific to your setup could be any of a million different problems. As I said above the travis file shows how QGC builds and android version on every pull request.