mavlink/qgroundcontrol #4867: New plan mission/save/sync model

AndKe, 2017-03-27:
In general, it looks great, a few issues need some attention:
1:
When connected, drawing / editing a plan have has way to abort. 
-Connect, plane is loaded from AP
-move some waypoint, change an altitude.
-observe that there is no way to re-read from AP, once you go back to "fly" view, the modified mission is uploaded.
2:
"Clear mission"  button is the only way to "close" an mission you opened, sounds like "remove all mission items" , please rename to "close mission" or something else.
3:
paralell saving of QGC WPL 110 formatted missins as .txt is needed . 
The .mission format is not "future-proof" until at least a few GCS's use that format.
Users may have xx older missions that are recurring annually, currently there is no way of loading them directly.  This may delay the adaptation of QGC.

DonLakeFlyer, 2017-03-27:
Closing since moved to #4868. 3 is already in another issue and I'm still trying to get it into 3.2.

Also keep in mind that my goal with the new json mission file format isn't to create something that is easy for other gcs to adopt. Nor am I concerned whether they do or not. QGC mission support is moving quickly beyond what anyone else has, hence it will be major work for them to support the new format and QGC featues (multiple complex item types, combined mission/fence/rally in same file, ...). I'd be surprised if anyone does that.

closed 2017-03-27 by DonLakeFlyer

DonLakeFlyer, 2017-03-27:
And thanks a ton for the feedback. Hugely helpful to have real people trying this stuff out.