mavlink/qgroundcontrol #4868: Plan: Problems with new sync user model

DonLakeFlyer, 2017-03-27:
* If the vehicle is flying it still blasts the new mission to the vehicle without warning
* No indication that the mission is being set to the vehicle
* No way to load files outside the normal path

DonLakeFlyer, 2017-03-27:
From @AndKe:
When connected, drawing / editing a plan have has way to abort.
-Connect, plane is loaded from AP
-move some waypoint, change an altitude.
-observe that there is no way to re-read from AP, once you go back to "fly" view, the modified mission is uploaded.

I'm not quite sure what you mean here. At that point both the Plan and Vehicle have the same misison?

AndKe, 2017-03-27:
Imagine this: 
-Connected to plane
-loaded the onboard mission, you are looking at current mission.
-by accident, some waypoints are moved, changed,  - if you exit "plan view" - the changes are uploaded to plane.
or: if you have changed some waypoints, you just wish to revert to the mission that is in the autopilot.

in short; I see no option to make changes and not upload to connected vehicle from "planning mode".

DonLakeFlyer, 2017-03-27:
> If the vehicle is flying it still blasts the new mission to the vehicle without warning

Ok, got it. Thats what I mean by the one above.

DonLakeFlyer, 2017-03-27:
> Thats what I mean by the one above.

Well only sort of. That case is for flying, but I understand what you mean.

AndKe, 2017-03-27:
Any mission upload should be done on request only, not just because it's critical, but it's seldom a success, and often just overload the link if the telemetry is slow/relayed or Iridium

LorenzMeier, 2017-03-27:
@AndKe The main use cases have no issues uploading a mission. The long-tail use cases with weak links should not dictate the user model for the 99% of users. Instead it should have special-case handling (e.g. via settings).

Anybody with a 3DR radio or better (which is pretty much everybody these days) benefits from less confusing mission handling.

@DonLakeFlyer There is one issue with the new model though: The upload is implicit. I don't think it should be implicit. It should be very clear and the user should be *asked if he wants to upload it*. I also think that on top of that a visual cue is required that clicking fly will upload it.

Right now if I assume that I have to upload I will search the screen for 5 minutes looking for the upload button. Even if the upload is automatic something in the UI should tell me which direction I need to go to benefit from that automation.

DonLakeFlyer, 2017-03-27:
I hear what each of you is saying. Still a WIP.

@AndKe I'm not going to bias for Iridium at this point. There are so many other problems with high latency they need to be dealt with all together (there is an issue in for that already). Unlikely for 3.2.

NaterGator, 2017-03-27:
I'm in agreement with @AndKe here; there is an inherent disconnect from pilot intention with this change. I'm not actually against it (seems like a useful feature that may decrease cognitive load for missions/use-cases where it fits), but I think the approach to transitioning to it as a default and the option to "revert" to pilot initiated mission transfers should still exist. Especially given that https://github.com/PX4/Firmware/issues/5130 still needs to be worked out, the failure-mode on the PX4 side when it rejects a mission upload is not particularly flexible for fixedwing.

DonLakeFlyer, 2017-03-27:
WIP...

AndKe, 2017-03-27:
@DonLakeFlyer , Yes Iridium is an less common , and bad example. let me use a more common example:
if people are flying with 3dr radios (MH-TRP) longer range, bigger missions, - syncing an 80 item mission may take minutes. - simply due to drops and retry, and temporary loss in sharp turns etc.

DonLakeFlyer, 2017-03-27:
If you know you upload when you go back to Fly. Is there a reason to go back to Fly view before you are done editing a mission?

LorenzMeier, 2017-03-27:
Plus the view switch will (should not) trigger the sync. Its if you *change* the mission. At which point you need to resolve the difference to the one onboard somehow.

LorenzMeier, 2017-03-27:
None the less your long range example is still outside of what ~95% of users do. So it again might need its own special mode, but should by no means dictate the user model of QGC.

DonLakeFlyer, 2017-03-27:
Not quite what I was looking for. But I think I have enough now to be dangerous.

DonLakeFlyer, 2017-03-27:
Ok, I have a new pull. I'm not explaining how it works now. Because I'd like you guys to try a daily build once it's done and see if you can figure it out yourself. If you can't that's a problem in itself. I'll tell you when a new daily is ready.

NaterGator, 2017-03-27:
Sounds good Don, we'll give it a shot.

DonLakeFlyer, 2017-03-27:
I will say one thing, which is that I'm realizing already the "Clear" concept is still broken. But I'm going to go out and fly this myself once the build finishes before I make any more changes.

DonLakeFlyer, 2017-03-27:
Ok. All new daily builds expcept for OSX are posted. Heading out to fly...

AndKe, 2017-03-27:
the checkbox fixes the basic need just fine.

Antiheavy, 2017-03-27:
I think this new user model is an interesting approach.   I just installed the recent daily with the Mission Settings "automatically upload on exit" checkbox.  I didn't mean for this post to be so long, but it it sort of grew as I've been trying this in fixed wing HIL:

1) Burying the [Upload] button inside the Mission Settings dialog may not scratch the itch for people that want an explicit [Upload] or [Sync] button.  The flashing [Upload] button in the upper right of the screen (found in daily builds from a few days ago) was really good for this.  Maybe put both the checkbox option and flashing [Upload] button it up there in upper right toolbar area to be always seen?

2) I don't understand the [Close] button in the mission settings...

3) This user model highlights a issue with PX4/QGC in that when a mission is uploaded during flight the mission index is reset to the beginning.  Just switching back and forth between Plan and Fly keeps resetting the mission to the beginning - super annoying.  Ideally QGC and/or PX4 would do the following:
i) know if anything really changed or different from what is on the vehicle and only prompt/auto upload if there is a change/difference.
ii) be smarter about choosing the active waypoint in a mission uploaded during flight (lots of possible approaches here).

4) This user model highlights some problems with PX4 MissionFeasibilityChecker:
i) Switching to Fly view will upload the mission, but the mission may be rejected by PX4.  It looks to the user as if there are no problems in Fly view when in fact there is a problem.
ii) PX4/Firmware#5130 is about trying to fix part of this problem.

5) What happened to Disarm button after landing?  Or is it just the way my HIL is behaving?  Does this have something to do with "No action after mission end" option in Mission Settings?

6) Mission Settings pane is becoming pretty confusing as more and more is being stuffed in there.  Also, the pane part of a list of waypoints, but it itself is not a waypoint which is a little weird to start off with.

still testing things.....

DonLakeFlyer, 2017-03-27:
> The flashing [Upload] button in the upper right of the screen (found in daily builds from a few days ago) was really good for this.

Makes sense.

> I don't understand the [Close] button in the mission settings

Closes the currently open mission so you can start creating a new one.

@LorenzMeier Could we do these new things behind a new MISSION_V2 capability bit:
* Add a new "Mode" to mission upload where the firmware doesn't pay attention to the current item specified in the mission item. But instead uses the one it already has with the new mission so it can keep on going?
* Add a new ack return for upload which indicates upload success but feasibility check failure with text?

DonLakeFlyer, 2017-03-27:
> know if anything really changed or different from what is on the vehicle and only prompt/auto upload if there is a change/difference.

Bug.

Antiheavy, 2017-03-27:
> Closes the currently open mission so you can start creating a new one.

how is this different from [Clear]?  from a user perspective it is unclear (hey, a pun!).

DonLakeFlyer, 2017-03-27:
Like I said "Clear" is kind of out of whack.

DonLakeFlyer, 2017-03-27:
New pull:
* Upload button is back in toolbar. Only shows up if there is something to upload.
* "Clear" changed to "Remove All" which will remove all items from the mission.
* "Close" changed to "New Mission"
* Fixed dirty bit checked. Mission will only upload if dirty.
* If you try to upload a mission to a vehicle which is armed you are prompted to make sure that is what you want to do. This happen whether you are using manual or auto-magic upload.

Antiheavy, 2017-03-28:
"Start Mission" button seems a little weird, like two stages of it?  The Arm, Disarm buttons are gone/missing.  not sure how to disarm at the end of flight nor re-arm for another flight.

DonLakeFlyer, 2017-03-28:
That's not associated with this item. But anyway:
* Start will arm and then start the mission automatically
* Disarm missing is a bug. I forgot to put that one in. The bug is that it should check for whether auto-disarm is turned on. If it isn't then it should be there. If auto-disarm is on, then a Disarm button is not needed.

DonLakeFlyer, 2017-03-28:
I've been playing around with the new Plan stuff a bunch. It's closer but still have a bunch of work to make it real. It's still pretty confusing. Mostly problems around:
* Clearing the mission from the vehicle
* If a mission is loaded, how to do change it completely. The whole New Mission versus loading another mission is confusing. Also @Antiheavy comment on Mission Setting getting cluttered. 

Will put some more time into this tomorrow.

DonLakeFlyer, 2017-03-28:
I've decided that my trying to implement the concept of auto-save and auto-sync for missions both at the same time just wasn't working. I think it's doable but it will require way more work than I have time right now. So instead I dropped auto-save and kept auto-send. Due to the various quirks described above I decided to default auto-sync to off for now.

The sync dropdown is now back. Upload in the toolbar will show up if you have auto sync off. The auto-sync setting is in the sync dropdown. If you turn on auto-sync the mission will be sent to the vehicle when you leave the plan view if dirty.

Also if the vehicle is armed and is currently flying a mission you are not allowed to upload a new misison since it would trash the current mission.

LorenzMeier, 2017-03-28:
Don, are you checking the disarm on land param for the disarm button? If you don't, please add. That would fix the missing button here.

DonLakeFlyer, 2017-03-28:
I know haven't done it yet. It's on my list.

LorenzMeier, 2017-03-28:
@DonLakeFlyer The requirement to not be in mission for upload is wrong for PX4. It supports mission changes on the fly and we don't want that limitation for our user base.

DonLakeFlyer, 2017-03-29:
It's going to reset the current mission item to the first item. Is that really what you want? If you really want to fix that then I think it requires mavlink protocol changes I suggested above.

dagar, 2017-03-29:
Why does it require mavlink protocol changes? Can't you set current=true in only the appropriate MISSION_ITEM? http://mavlink.org/messages/common#MISSION_ITEM

DonLakeFlyer, 2017-03-29:
Two things:
* Seems like the ground station could have stale data on current waypoint at edge cases. Hence causing the vehicle to return to an old point.
* The ground station has no way of knowing that the mission that is in the Plan view is the same as the mission being flown.

I can disregard both of these and do what @dagar says.

dagar, 2017-03-29:
Right, I don't think it's a good idea unless we're confident the current mission is in sync with the planned.

Then it seems like you'd want the mission to usually reset to 0 except for a few special cases.
 - if you changed a waypoint index >= current active
 - if the waypoint type didn't change
 - if the changed waypoint is a regular position or loiter

Then you could get the desired behaviour of tweaking an active mission. I don't know if this is realistic with the current QGC mission planning.

dagar, 2017-03-29:
Could we do a better job with partial sync? Establish a simple mission checksum, quickly verify both sides match, then only sync the changed item? MISSION_WRITE_PARTIAL_LIST?

DonLakeFlyer, 2017-03-29:
Let me do some poking around to see how other ground stations handle this. I'm going to close this issue and make a new one just for this.

DonLakeFlyer, 2017-04-04:
This is all closed out now. I'll deal with individual complaints in new Issues.

closed 2017-04-04 by DonLakeFlyer