mavlink/qgroundcontrol #4876: Nested parameters

dagar, 2017-03-28:
@DonLakeFlyer - I was thinking of expanding and better organizing the PX4 parameter groups. Any thoughts on how we could adding nesting? Another metadata field, or do it automatically based on name?

For example I wanted to split `VTOL Attitude Control` into 
 - VTOL
   - Standard
   - Tailsitter
   - Tiltrotor

DonLakeFlyer, 2017-03-29:
I'm not convinced another level of hierachy is the way to go here. Concerned it just adds more complexity to an already complex ui.

dagar, 2017-03-29:
One motivation for this is the VTOL group where parameters for 3 mutually exclusive VTOL types are mixed together with no distinct naming. Another is separating the parameters you currently need to change to make the system usable from the parameters that should rarely ever be changed. In general I want to push for fewer params, but that's another discussion.

For example adding "FW Attitude Control Advanced" within "FW Attitude Control".

FW Attitude Control
  - Advanced
    - advanced param 1
    - advanced param 1
  - param1
  - param2

I think we should try and get to the point a regular user never has to see the parameter tree at all, but right now it's unavoidable and becoming a confusing mess.

DonLakeFlyer, 2018-01-03:
https://github.com/PX4/Firmware/issues/8568

DonLakeFlyer, 2018-01-04:
Replaced by #5965

closed 2018-01-04 by DonLakeFlyer