mavlink/qgroundcontrol #4882: Why are daily builds tagged as 3.1.2?

abencomo, 2017-03-29:
The latest [QGC release is 3.1.3](https://github.com/mavlink/qgroundcontrol/releases/tag/v3.1.3).  However, the [daily builds](https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/download_and_install.html) are labeled 3.1.2. Why is that?

DonLakeFlyer, 2017-03-29:
Where are you seeing a version number on daily builds? The title should just be a git commit number on daily builds.

abencomo, 2017-03-29:
The first screenshot shows 3.1.2 a the version for the daily build I just downloaded today.  The second one is the official release build, and it says version 3.1.3 in the Info screen.  However, the top title does show the git commit number as you pointed out so I guess is not a big deal.  I close this.

[image]
[image]

[image]

DonLakeFlyer, 2017-03-29:
Ah, I see. Just a build wierdness. Going to re-open since worthwhile fixing at some point.

closed 2024-06-24 by DonLakeFlyer