mavlink/qgroundcontrol #4902: Survey: YAW is defaulting to North

tops4u, 2017-04-02:
When setting MIS_YAWMODE=0 (as defined by Waypoint) and using Survey with a non default Grid Angle the MC will always use YAW=0°. 
My expectation was that if I set a specific Grid Angle that it would use this for YAW. 
Currently I have IMHO no way to fly a Survey with a fixed YAW Angle for the whole Survey.

For all other Settings of MIS_YAWMODE the YAW Angle of the Waypoint should have not influence, so SURVEY should IMHO take YAW as specified from the Survey Grid Angle.

DonLakeFlyer, 2017-04-02:
Survey does not pay any attention to mission yaw parameters. You are expected to set that up correctly. For Survey you should be using MIS_YAWMODE=1 so that the vehicle always points in the direction of flight. Is there a problem with using MIS_YAW_MODE=1?

tops4u, 2017-04-02:
Yes. We intend to have YAW in a fixed direction for the whole mission. Currently this is impossible. Would it be possible to habe Survey take over the Grid Angle. In any other Setting of MIS_YAWMODE this would have no Impact.

DonLakeFlyer, 2017-04-02:
You are going to need to handle this manually. The way to do it is to set a waypoint before the mission with the heading set to what you want. Then set the parameter to not change yaw. The problem is that as far as I currently know. Neither PX4 Pro or ArduPilot firmware support yaw in a waypoint mission command item. 

I don't see adding this as a feature to survey unless it becomes a common request (and the firmwares support it). Survey is already complicated enough with all the settings.

If you find that either of the firmwares supports setting yaw in a waypoint command then let me know and I'll update QGC to allow for that. If not, I'd ask them to start supporting it and once again I'll update to that new support.

tops4u, 2017-04-02:
@DonLakeFlyer How am I going to handle this manually? Does Survey not _simply_ translate the Survey Area into Waypoints - where "simply" refers to my understanding. 
When I set YAW for PX4 Waypoints this works perfectly well until I reach Survey where the UAV will turn YAW to north and continue the Survey with a North Heading. So I thought that when Survey creates the Waypoints I could simply apply the GRID Angle as YAW Setpoint!? 

Am I misstaken?

DonLakeFlyer, 2017-04-02:
Hmm, I'll have to do some poking around the firmware code.

DonLakeFlyer, 2017-04-02:
What I've discovered so far:
* PX4 Pro supports Yaw in waypoint
* ArduPilot does not

@LorenzMeier There isn't some way to send a NAV_WAYPOINT mission item such that it disregards the yaw value is there?

dagar, 2017-04-02:
Use NaN?

https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_mission.cpp#L951

DonLakeFlyer, 2017-04-02:
I tried to look for that but couldn't find any special processing for NaN on yaw.

dagar, 2017-04-02:
It should be preserved all the way through mission handling.

DonLakeFlyer, 2017-04-04:
Talked with Lorenz. Vehicle will respect NaN on yaw. So that is how to fix this. Two changed needed:
1) Survey needs to set NaN for yaw on all it's waypoint
2) Mission items which support yaw must be correctly marked as supporting NaN. Then item editors need to show Yaw with checkbox as well as edit field.

DonLakeFlyer, 2017-04-04:
Also need to update mavlink spec.

tops4u, 2017-04-04:
Huh... this seems rather extensive. Sorry for this one.

DonLakeFlyer, 2017-04-04:
Not really. This current thing is just wrong.

closed 2017-04-08 by DonLakeFlyer

tops4u, 2017-05-04:
@DonLakeFlyer I have seen this is fixed. I was wondering how to set the YAW for a certain Waypoint? I can't see this Option anymore. Could you help me out?

DonLakeFlyer, 2017-05-05:
If you are planning a mission while disconnected from a vehicle it should always show up for a PX4 Multi-Rotor in mission settings. If you are connected to a vehicle you have to have MIS_YAWMODE set to the value which allows you to specify yaw to get it to show up.

tops4u, 2017-05-05:
@DonLakeFlyer  this worked. I was wondering why the YAW Field is not shown while disconnected. AFAIR there was the possibility to choose for which Firmware Type one wanted to plan Missions. This is rather inconvenient when planing Missions ahead (for example at home when you plan the flight without having the Vehicle at hand).

DonLakeFlyer, 2017-05-06:
> AFAIR there was the possibility to choose for which Firmware Type one wanted to plan Missions

First item in Plan, Mission Settings - Vehicle Info. You can only set this when there are no items in the mission.