mavlink/qgroundcontrol #4906: Standard altitude and VTOL commands altitude

sanderux, 2017-04-02:
It would be nice to be able to set the default altitude.
Further more the VTOL_TAKEOFF and VTOL_LAND commands always seem to set 50m as altitude, it would be better to have them follow the standard waypoint behavior (as previous waypoint)

Finally it seems the previous altitude only seems to be stored when the input field looses focus, if the next action is a click on the map to set a new waypoint while the cursor was still in the altitude input box it will set the next waypoint with a different altitude.

DonLakeFlyer, 2017-05-04:
* VTOL_TAKEOFF now uses default altitude
* VTOL_LAND sets altitude to 0 just like regular NAV_LAND
* Input field focus problems fixed a while ago. Note: There still may be some holes here so please report problems if you are still seeing them.

closed 2017-05-04 by DonLakeFlyer

sanderux, 2017-05-04:
Hi Don,

The VTOL_LAND command, unlike the NAV_LAND is a waypoint command.
I believe it is wrong to assume 0 altitude. Because this was standard behavior in QGC we currently disregard the altitude in this command but at a certain time we should start using altitude as especially for VTOL land people might want to change the altitude the vehicle flies to before initiating landing.

DonLakeFlyer, 2017-05-04:
Why isn't the altitude value in VTOL_LAND the ground height you want to land at like it is for regular _LAND? Hence the 0 default.