mavlink/qgroundcontrol #4908: Help with parameter update in QGC

benguru, 2017-04-03:
I have an issue with the parameters appearing only in the default group.
I have tried many times the custom firmware upload through QGC but without success.
I am using QGC 3.1.3

Here is an extract of the file ParameterFactMetaData.xml and the parameters show here but not in the GUI:
`<group name="Bifoiler Control">
    <parameter default="0.4" name="BF_L_TC" type="FLOAT">
      <short_desc>Lift control Time Constant</short_desc>
      <long_desc>This defines the latency between a speed step input and the achieved setpoint (inverse to a P gain). Half a second is a good start value and fits for most average systems. Smaller systems may require smaller values, but as this will wear out servos faster, the value should only be decreased as needed.</long_desc>
      <min>0.4</min>
      <max>1.0</max>
      <unit>s</unit>
      <decimal>2</decimal>
      <increment>0.05</increment>
      <scope>modules/fw_att_control</scope>
    </parameter>`

Any suggestion, do I need to modify something else in the px4 firmware ?

DonLakeFlyer, 2017-04-04:
You need to bump the version number.

benguru, 2017-04-05:
this is done in the xmlout.py right ?
I am new to this I am sorry if my questions sounds foolish...
Do I have to change the xml_version (in the code snippet down below ?)
If not, how and in which file do you do that.

xml_parameters = ET.Element("parameters")
        xml_version = ET.SubElement(xml_parameters, "version")
        xml_version.text = "3"
        xml_version = ET.SubElement(xml_parameters, "parameter_version_major")
        xml_version.text = "1"
        xml_version = ET.SubElement(xml_parameters, "parameter_version_minor")
        xml_version.text = "10"
        importtree = ET.parse(inject_xml_file_name)

DonLakeFlyer, 2017-04-05:
If I remember correctly, yes

closed 2017-07-22 by benguru