mavlink/qgroundcontrol #4934: QGC fails to run - RunGuard fails, need help!!

samacumen, 2017-04-07:
I am on latest qgc master, and QGC is unable to start. But, it might be a local machine issue, please guide. 

Starting /build-qgroundcontrol-Desktop_Qt_5_8_0_GCC_64bit-Debug/debug/QGroundControl...
QML debugging is enabled. Only use this in a safe environment.
/build-qgroundcontrol-Desktop_Qt_5_8_0_GCC_64bit-Debug/debug/QGroundControl exited with code 0

```
int main(int argc, char *argv[])
{
#ifndef __mobile__
    RunGuard guard("QGroundControlRunGuardKey");
    if (!guard.tryToRun()) {
        return 0;
    }

// I saw that the reason is, in tryToRun this line returns false.
const bool result = sharedMem.create( sizeof( quint64 ) );
```

My System configurations:
QtCreator 4.2.1 (Qt 5.8), Ubuntu 16.04 LTS, Dell XPS15, 16GB RAM

I have never had problem like this before. Any help is appreciated.

DonLakeFlyer, 2017-04-07:
I would ask on Gitter there are more Linux folks there. Issues are just for bugs.

closed 2017-04-07 by DonLakeFlyer

furkantektas, 2017-12-30:
This may not be a Linux bug. I've encounter the exact same situation on different machines and different versions of QGroundControl (3.2.x). While `RunGuard::isAnotherRunning()` function returns false,  `RunGuard::tryToRun()` returns also false as function fails to create a shared memory instance. In this case, one should detach from the shared memory and re-attach to solve this issue. 

I've modified `tryToRun` function as follows to avoid this bug:

```
bool RunGuard::tryToRun()
{
    if ( isAnotherRunning() )   // Extra check
        return false;

    memLock.acquire();
    bool result = sharedMem.create( sizeof( quint64 ) );
    memLock.release();
    if ( !result )
    {
        sharedMem.detach();
        bool result = sharedMem.create( sizeof( quint64 ) );
        if ( ! result ) {
            release();
            return false;
        }
    }

    return true;
}
```