mavlink/qgroundcontrol #4935: Fixed wing Abort landing button is buried, too many clicks

Antiheavy, 2017-04-07:
In recent daily builds the [Abort] landing button is buried underneath the new "Action" context button.  It now requires a click on one part of the screen, move the mouse to another part of screen, click, then slide to accept.  [Abort] needs to be much quicker than that: one click, or one click + slide to accept at most.

Also the [Abort] button shows up in wrong contexts.  It should only be shown when:

-  the aircraft sends some status message that it is in an "isLanding" state.  PX4 does an abort to loiter over the active land waypoint.  There are other "landing" modes/states that PX4 could be in do not use a "Land" waypoint; in these cases the [Abort] button may not be appropriate to display - might this be problematic?

- QGC detects a "Land" type waypoint is active.  Is this possible?  If so might this be better?  It would not require anything additional from the firmware and it would ensure that a valid landing point was available to loiter over.

reference:
https://github.com/PX4/Firmware/pull/6539
https://github.com/mavlink/qgroundcontrol/issues/3725

DonLakeFlyer, 2017-04-07:
@dagar Can you get me a landing state in EXTENDED_STATUS?

dagar, 2017-04-10:
I've asked this a few different times in different issues and haven't gotten an answer. Can we extend enums in mavlink?

DonLakeFlyer, 2017-04-10:
I don't see why not. Shouldn't affect existing code in a bad way.

dagar, 2017-04-12:
@DonLakeFlyer please review https://github.com/mavlink/mavlink/pull/698 and I'll add firmware support.

DonLakeFlyer, 2017-05-08:
Fixed

closed 2017-05-08 by DonLakeFlyer