mavlink/qgroundcontrol #4949: Fly Screen - cosmetic issue

lvale, 2017-04-09:
When flying a mission the red line behaves like the snake game, ie like a trail but dies not extend to all the previous points

Moment 1
[image]

Moment 2
[image]

If the intention of the red line is to indicate what has been done, shouldn't it be from where the path began instead of "following" ?

DonLakeFlyer, 2017-04-09:
There is a limit on the length of the trail due to performance.

DonLakeFlyer, 2017-04-09:
This was mainly to fix a perf problem on mobile. I'll take a look at it again to possibly set longer trails on desktop.

closed 2017-05-08 by DonLakeFlyer