mavlink/qgroundcontrol #4950: Mission Planning - Economic route

lvale, 2017-04-09:
On this test, I drew a mission and then checked the option for the 90 degrees.

The resulting mission is not very "economic".

See the case of WP27,28,39 where the vehicle will have to cross the entire survey grid, to then do the 90 degress pattern, and end the mission at the furthest of home.

[image]

[test_mission1.mission.txt](https://github.com/mavlink/qgroundcontrol/files/908142/test_mission1.mission.txt)

IMHO, would be better to start the 90 degrees at the bottom and work the way up, and that would result in a more "economic" route.

Perhaps, also, adding, an option for "entry" and "exit" points would be helpful.

lvale, 2017-04-09:
Also. Shouldn't WP28 be more "close" to WP3 ?

Antiheavy, 2017-04-09:
have you tried 270 degrees instead of 90 degrees?  sometimes that helps put the start and end points at better locations.  I don't know if that fixes your problem in this instance or not.

DonLakeFlyer, 2017-04-09:
Yup, known problem. Not fixed yet.

DonLakeFlyer, 2017-04-14:
> Also. Shouldn't WP28 be more "close" to WP3 ?

Not really. It looks like it is at the right point. Which you can verify that the distance from the transect to waypoint 16 equals half the distance between transects. So it's in the right place to get the coverage needed.

DonLakeFlyer, 2017-05-04:
Hover and Capture will now connect the refly section using the most economic route.

closed 2017-05-04 by DonLakeFlyer