mavlink/qgroundcontrol #4973: Add support for testing PX4 camera trigger

mhkabir, 2017-04-13:
@dogmaphobic @LorenzMeier 

For users to test PX4 camera triggering, we need to send a `DO_DIGICAM_CONTROL` () command, with Param 5 > 0. It would go on to the existing "Camera" page in QGC as a button, which would fire the trigger one-off.

[image]

There is a pending pull request (https://github.com/PX4/Firmware/pull/6973) which fixes triggering and makes sure that the command is handled properly.

DonLakeFlyer, 2017-04-13:
It's already there. Swithch to the camera page in the center instrument panel.

mhkabir, 2017-04-13:
[image]
Where?

DonLakeFlyer, 2017-04-13:
> Swithch to the camera page in the center instrument panel.

DonLakeFlyer, 2017-04-13:
> Swithch to the camera page in the center instrument panel.

[image]

But now I see you want it on the setup page as well. Makes sense.

mhkabir, 2017-04-13:
Thanks! 

Yes, it would make sense to have it in setup. I don't think that anyone would want to do one-off triggering from the instrument cluster.

mhkabir, 2017-04-13:
@DonLakeFlyer I can't find the camera page in the center instrument panel. Is that in master? I'm running the latest daily.

[image]

DonLakeFlyer, 2017-04-13:
It's been in master for about a month now

mhkabir, 2017-04-13:
Maybe it's only enabled for APM ?

DonLakeFlyer, 2017-04-13:
Nope. You aren't running the latest osx daily build. If you were it would crash, since it's broken right now.

mhkabir, 2017-04-13:
Eugh, sorry about that. Didn't read :P

mhkabir, 2017-04-21:
@DonLakeFlyer The "trigger camera" button does not seem to be actually sending anything to the vehicle. I'm checking for commands on the PX4 side, but no DIGICAM_CONTROL command comes through.

LorenzMeier, 2017-04-23:
@mhkabir Your testing must be bad. It does arrive in Firmware.

LorenzMeier, 2017-04-23:
The command sent is this: ```    sendMavCommand(FactSystem::defaultComponentId,
                   MAV_CMD_DO_DIGICAM_CONTROL,
                   true,                            // show errors
                   0.0, 0.0, 0.0, 0.0,              // param 1-4 unused
                   1.0);                            // trigger camera```

LorenzMeier, 2017-04-23:
The root cause is this issue: https://github.com/mavlink/qgroundcontrol/issues/5045

DonLakeFlyer, 2017-04-23:
Bad component id for command fixed here: https://github.com/mavlink/qgroundcontrol/pull/5048

DonLakeFlyer, 2017-05-08:
@mhkabir This setup page is pretty out of whack. The TRIG_INTERFACE parameter doesn't seem to been the latest master firmware. Do you know what is going on there?

mhkabir, 2017-05-08:
Hi @DonLakeFlyer It should be working just fine. I tested it multiple times over the last few days. 

Just in case, please use this branch, which includes many fixes : https://github.com/PX4/Firmware/pull/7201

closed 2017-05-21 by DonLakeFlyer