mavlink/qgroundcontrol #4974: The latest daily build doesn't work on Mac OS X 10.10.5 anymore

IeiuniumLux, 2017-04-13:
I donwloaded today's daily build and I got this message when I tried to launch it.  The last daily build I used from last week works fine.
[image]
[image]

dogmaphobic, 2017-04-13:
Installer issues? It's not finding the Qt libraries. Specifically, QtLocation.

DonLakeFlyer, 2017-04-13:
Hm, thought I had fixed that. I'll look.

IeiuniumLux, 2017-04-13:
Let me know if you want me to test/confirm the fix once you have a daily build available for it.

DonLakeFlyer, 2017-04-13:
Usually means I pulled something out of the Qt 5.7 image up on S3 that I shouldn't have.

DonLakeFlyer, 2017-04-14:
Fixed

IeiuniumLux, 2017-04-14:
Is there a new daily-build or do you have wait until tomorrow?

DonLakeFlyer, 2017-04-14:
New daily is posted.

IeiuniumLux, 2017-04-14:
Unless there is a different url for the daily builds, then it still doesn't work.  I just downloaded it from here:
https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.dmg

DonLakeFlyer, 2017-04-14:
Hmm, strange. It works fine on one of my machines but not on the other.

DonLakeFlyer, 2017-04-14:
@dogmaphobic Do you have some quick time to look at this. The strange thing is that the QtLocation framework is in the package if you open it up? Not sure what is going wrong in there.

DonLakeFlyer, 2017-04-14:
I thought that the travis s3 Qt tar was wrong. But looks like that wasn't it.

DonLakeFlyer, 2017-04-14:
If I build locally using the same tarball it wll works fine. Hmm... A bit stumped at this point.

DonLakeFlyer, 2017-04-14:
Ok, so I just tar'ed up a full regular 5.7 install. That didn't work either. Something strange going on with 5.7.1.

DonLakeFlyer, 2017-04-14:
@dogmaphobic I have no idea what is going on here with QtLocation. Seems to have broken with 5.7.1.

dogmaphobic, 2017-04-14:
I downloaded the current daily and it booted fine. How do I reproduce this?

DonLakeFlyer, 2017-04-14:
Which is kind of what I've been seeing as well. Sometimes it works and sometimes it doesn't. I change something, test it. It all works fine. Download it. It all works fine. Then someone reports it not work, I download again. And suddenly the same exact things doesnt' work.

DonLakeFlyer, 2017-04-14:
And I just tried it again and it worked. WTF!

@IeiuniumLux Can you try?

mhkabir, 2017-04-16:
@DonLakeFlyer works for me now.

DonLakeFlyer, 2017-04-16:
Great thanks. Maybe it's fixed maybe it's not. Not srue what caused it in the first place and not sure why it went way. Fingers crossed.

closed 2017-04-16 by DonLakeFlyer