mavlink/qgroundcontrol #4988: MAV_CMD(0) command failed

baronep, 2017-04-15:
I am working on implementing a custom MAVLINK interface, but every time I connect my flight controller, this message appears 
```
MAV_CMD(0) command failed
```

As far as I can tell, QGroundcontrol is sending a COMMAND_LONG message, with a command=0 field. The command field should map to the MAV_CMD enum, but there is no entry for 0 in the MAV_CMD enum. Can someone explain why this is happening?

DonLakeFlyer, 2017-04-15:
Are you using a modified verison of QGC? If not I doubt that QGC is sending a bad command to your vehicle. If it was everyone using QGC would be reporting this problem. Yet there have been no reports of such a thing. My guess it that your vehicle is sending back an incorrect COMMAND_ACK message. Which QGC is then display a message about.

closed 2017-04-15 by DonLakeFlyer

baronep, 2017-04-17:
@DonLakeFlyer

I have done a little more investigating into the issue, and I am in fact receiving a COMMAND_LONG message with command=0. The catch is that I was not yet checking the target_system and target_component of the command, both of which were set to 0. 

The reason that the PX4 flight stack does not respond is because it ignores any mavlink commands where the sysid and compid do not match their system's (with the exception of the MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES for some reason).  (Referencing the PX4 flight stack: Firmware/src/modules/mavlink/mavlink_receiver.cpp:348).

So it looks like QGC is sending the message, but other autopilots correctly ignore the message.

DonLakeFlyer, 2017-04-17:
I'll poke around to find it