mavlink/qgroundcontrol #4996: Survey tool polygon glitch

feemi, 2017-04-16:
[image]

When randomly editing polygon edges. Survey tool is failed.

DonLakeFlyer, 2017-04-16:
I've been chasing that one for a while. Would be awesome if you could figure out a set of concrete repro steps for it.

DonLakeFlyer, 2017-04-16:
Do you remember if you were just moving edges? Or if you also zoomed the map as well.

tops4u, 2017-05-29:
I have had this Problem too recently. AFAIR this happened while paning and zooming around.

DonLakeFlyer, 2017-08-12:
Closing since there are no concrete steps to reproduce. I believe there is a Qt bug somewhere causing this. Fingers crossed that 3.3 moe to 5.9 will fix.

closed 2017-08-12 by DonLakeFlyer