mavlink/qgroundcontrol #4998: QGC checking for parameters that aren't set in HIL model

Antiheavy, 2017-04-16:
I get a lot of param/sensor errors thrown in QGC when HIL testing recent PX4 Master firmware.  This shows up with the PX4 HILstar(Xplane) model and Xplane 10 HIL sim.  It does not show up with when connected to real vehicles.

[image]

I don't totally understand the root cause, but @dagar  suggested I post this issue here.  https://github.com/PX4/Firmware/issues/7050#issuecomment-294352926

Any ideas?

DonLakeFlyer, 2017-04-16:
This is the firmware's concept of optional parameters. Which is a losing battle I continually fight with. QGC never expects these parameters to be missing. Hence it complains if they are because if thinks something is wrong.

dagar, 2017-04-16:
In this case can't we use the Mavlink spec instead of looking for specific parameters?

SYS_STATUS and MAV_SYS_STATUS_SENSOR
http://mavlink.org/messages/common#SYS_STATUS

Alternatively, HIL is a pretty special case where the sensors actually come in through QGC. It doesn't seem that unreasonable to skip the sensors component with HIL enabled.

DonLakeFlyer, 2017-04-16:
I"m pretty sure this will happen before you start hil. This is just connecting to an airframe. QGC doesn't know one airframe from another. No way to identify as far as I know.

dagar, 2017-04-16:
HIL is a MAV_MODE_FLAG in the HEARTBEAT base_mode, so hil mode is known on vehicle creation.

What about using SYS_STATUS? It appears to be implemented in PX4, and it looks like the devices themselves report their existence, rather than waiting for calibration to run and fill the CAL* params.

For now we can make an effort to keep this up to date. https://github.com/PX4/Firmware/blob/master/src/modules/sensors/parameters.cpp#L160

DonLakeFlyer, 2017-04-16:
> HIL is a MAV_MODE_FLAG in the HEARTBEAT base_mode,

Right but if I remember correctly you turn on HIL mode after you connect don't you? That's what the HIL widget thing does I thought.

That param list in the firmware doesn't work on HIL?

dagar, 2017-04-16:
It should be working, but doesn't have CAL_ACC0_ID and CAL_GYRO0_ID.

The airframe is a HIL type, and PX4 commander, sensors, etc startup knowing they're in HIL. I believe the QGC HIL widget starts the connection to xplane/flightgear and then starts streaming sensor data to the firmware.

DonLakeFlyer, 2017-04-16:
Vehicle doesn't start in HIL as far as I know. HIL mode is started through the HIL widget:
```
void Vehicle::setHilMode(bool hilMode)
{
    mavlink_message_t msg;

    uint8_t newBaseMode = _base_mode & ~MAV_MODE_FLAG_DECODE_POSITION_HIL;
    if (hilMode) {
        newBaseMode |= MAV_MODE_FLAG_HIL_ENABLED;
    }

    mavlink_msg_set_mode_pack_chan(_mavlink->getSystemId(),
                                   _mavlink->getComponentId(),
                                   priorityLink()->mavlinkChannel(),
                                   &msg,
                                   id(),
                                   newBaseMode,
                                   _custom_mode);
    sendMessageOnLink(priorityLink(), msg);
}
```

dagar, 2017-04-18:
A PX4 HIL airframe (SYS_AUTOSTART 1XXX) already has MAV_MODE_FLAG_HIL_ENABLED.

Antiheavy, 2019-08-17:
this issue is old and probably no longer applicable.  closing.

closed 2019-08-17 by Antiheavy