mavlink/qgroundcontrol #5031: Clone failing?

rhvonlehe, 2017-04-20:
I've tried multiple times to clone the QGC git repo and always get this:

```
H228253@MN10IWJ5XCDH2 MINGW64 /c/Projects/QGroundControl
$ git clone --recursive https://github.com/rhvonlehe/qgroundcontrol.git
Cloning into 'qgroundcontrol'...
remote: Counting objects: 82134, done.
remote: Compressing objects: 100% (101/101), done.
remote: Total 82134 (delta 39), reused 0 (delta 0), pack-reused 82033
Receiving objects: 100% (82134/82134), 192.74 MiB | 1.76 MiB/s, done.
Resolving deltas: 100% (62068/62068), done.
Checking out files: 100% (3355/3355), done.
Submodule 'libs/mavlink/include/mavlink/v1.0' (https://github.com/mavlink/c_library_v1.git) registered for path 'libs/mavlink/include/mavlink/v1.0'
Submodule 'libs/mavlink/include/mavlink/v2.0' (git://github.com/mavlink/c_library_v2.git) registered for path 'libs/mavlink/include/mavlink/v2.0'
Submodule 'src/GPS/Drivers' (https://github.com/PX4/GpsDrivers.git) registered for path 'src/GPS/Drivers'
Cloning into 'C:/Projects/QGroundControl/qgroundcontrol/libs/mavlink/include/mavlink/v1.0'...
remote: Counting objects: 10141, done.
remote: Compressing objects: 100% (214/214), done.
remote: Total 10141 (delta 172), reused 0 (delta 0), pack-reused 9927
Receiving objects: 100% (10141/10141), 4.26 MiB | 1.12 MiB/s, done.
Resolving deltas: 100% (9152/9152), done.
Cloning into 'C:/Projects/QGroundControl/qgroundcontrol/libs/mavlink/include/mavlink/v2.0'...
fatal: unable to connect to github.com:
github.com[0: 192.30.253.113]: errno=Invalid argument
github.com[1: 192.30.253.112]: errno=Invalid argument

fatal: clone of 'git://github.com/mavlink/c_library_v2.git' into submodule path 'C:/Projects/QGroundControl/qgroundcontrol/libs/mavlink/include/mavlink/v2.0' failed
Failed to clone 'libs/mavlink/include/mavlink/v2.0'. Retry scheduled
Cloning into 'C:/Projects/QGroundControl/qgroundcontrol/src/GPS/Drivers'...
remote: Counting objects: 254, done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 254 (delta 14), reused 0 (delta 0), pack-reused 231
Receiving objects: 100% (254/254), 92.46 KiB | 0 bytes/s, done.
Resolving deltas: 100% (182/182), done.
Cloning into 'C:/Projects/QGroundControl/qgroundcontrol/libs/mavlink/include/mavlink/v2.0'...
fatal: unable to connect to github.com:
github.com[0: 192.30.253.112]: errno=Invalid argument
github.com[1: 192.30.253.113]: errno=Invalid argument

fatal: clone of 'git://github.com/mavlink/c_library_v2.git' into submodule path 'C:/Projects/QGroundControl/qgroundcontrol/libs/mavlink/include/mavlink/v2.0' failed
Failed to clone 'libs/mavlink/include/mavlink/v2.0' a second time, aborting

H228253@MN10IWJ5XCDH2 MINGW64 /c/Projects/QGroundControl
```

I see no connectivity issues with Github or the internet.  I'm using other projects from Github on the same machine.  Any ideas?

rhvonlehe, 2017-04-20:
The url I'm cloning is a Github fork of the main repo.  I'm trying the main repo itself as well.

NaterGator, 2017-04-20:
Please run a `git submodule sync`

rhvonlehe, 2017-04-20:
Somebody else here at work figured out that our firewall doesn't allow git:// transfers and one or more of the submodules used git rather than https.  I can edit .gitmodules and fix it on my end.  It should eventually get fixed in the source.  I vote to make the transfer protocol consistent across all modules/submodules.

rhvonlehe, 2017-04-21:
Closed with recent pull request #5032

closed 2017-04-21 by rhvonlehe