mavlink/qgroundcontrol #5042: Error compiling GPS/GPSProvider.cc

TSC21, 2017-04-22:
I'm getting this error while compiling in `qtcreator`:
```make
compiling ../src/GPS/GPSProvider.cc
../src/GPS/GPSProvider.cc: In member function ‘virtual void GPSProvider::run()’:
../src/GPS/GPSProvider.cc:65:52: error: ‘GPSDriverUBX::Interface’ has not been declared
         gpsDriver = new GPSDriverUBX(GPSDriverUBX::Interface::UART, &callbackEntry, this, &_reportGpsPos, _pReportSatInfo);
                                                    ^
../src/GPS/GPSProvider.cc:66:20: error: ‘class GPSDriverUBX’ has no member named ‘setSurveyInSpecs’
         gpsDriver->setSurveyInSpecs(_surveyInAccMeters * 10000, _surveryInDurationSecs);
                    ^
Makefile:145046: recipe for target 'obj/GPSProvider.o' failed
make: *** [obj/GPSProvider.o] Error 1
18:15:39: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.7.1 GCC 64bit)
When executing step "Make"
```

DonLakeFlyer, 2017-04-22:
Your submodules are out of date.

closed 2017-04-22 by DonLakeFlyer

TSC21, 2017-04-22:
Thanks! Solved