mavlink/qgroundcontrol #5060: QGC survey generates too many WP's

lvale, 2017-04-26:
Test survey created
(latest today build)
[image]
[image]
[image]

This survey generates several thousand WP's

[t2.plan.txt](https://github.com/mavlink/qgroundcontrol/files/957008/t2.plan.txt)

DonLakeFlyer, 2017-04-26:
You have Hover and Capture turned on. That will generate waypoints to stop and take a picture at each interval.

> Also, the Fly screen should show the mission as "retrieved" from the vehicle and not from the generated Planning screen

Don't kow what you mean by that?

lvale, 2017-04-26:
On those screenshots above on the FLY screen the waypoints shown (the orange ones) are the ones generated by the Planning screen, and not the ones that the drone has (in this case it had none). If there is a mission on the drone that should be the one shown on the FLY screen.

DonLakeFlyer, 2017-04-26:
Can you give me the repro steps for that? Also I fixed a bunch of stuff around this two days ago. So make sure you are running latest.

lvale, 2017-04-26:
Just generate a mission like the one above. Try to upload it to vehicle. It throws out an error about too big. Switch to Fly screen and the waypoints are there, although none is on the vehicle.

DonLakeFlyer, 2017-04-26:
Oh, yeah. I had that one on my list of things to do. I'll fix it.

DonLakeFlyer, 2017-05-04:
Fixed

closed 2017-05-04 by DonLakeFlyer