mavlink/qgroundcontrol #5062: Altitude change pop-up very short range and heading random change

lvale, 2017-04-26:
Clicking on the Action to change altitude the elevator on the right only allows very small increments (10m ?)

[image]
[image]

Also changing altitude is changing also the heading for some undetermined heading

DonLakeFlyer, 2017-04-26:
Willing to consider new range. Any opinions?

lvale, 2017-04-26:
I would go for a log slider, meter by meter on the first half, then tens of meters, and maximum range below 400ft (120m), and/or the possibility of inputting a number.

On a usability, and I know it's not easy, we have to move from the extreme left to the bottom middle to the extreme right of the screen for a single process. Couldn't we have these elements to be closer from one another ?

DonLakeFlyer, 2017-04-26:
Log slider is way too much work for now. Also don't think I'll go back to allowing numeric entry. Just trying to close things out with major usability problems.

pritamghanghas, 2017-05-14:
heading change is Ardupilot problem. It doesn't honor SET_POSITION_TARGET_GLOBAL_INT yaw parameter. There are several holes in arducopter mavlink implimentation. It is like anytime I use a command something is not implimented especailly when you have anything to do with yaw. Another day I was trying to specify rate of yaw change in a yaw change command, arducopter ignores that as well.

amilcarlucas, 2017-05-14:
@pritamghanghas I did a couple of PRs for missing YAW implementations in Arducopter. Those PRs did not get accepted in the master branch, but if you apply them localy it might help you.

pritamghanghas, 2017-05-14:
@amilcarlucas thank you for that. I generally stay away from touching flight code. But it is a pity that those didn't get accepted. QGC can't really use something that is not in mainline. Though I can use them personally.

DonLakeFlyer, 2017-05-14:
I'm going to stay away from guided heading change for now. Depending on vehicle settings you can sort of achieve it with goto location. May consider at some point.

closed 2017-05-14 by DonLakeFlyer