mavlink/qgroundcontrol #5063: Takeoff action should allow height to takeoff

lvale, 2017-04-26:
I understand the current behaviour mimics what DJI does, but I believe it should allow to change the takeoff height
[image]

DonLakeFlyer, 2017-04-26:
And I've heard the exact opposite feedback to that as well. Trying to keep things as simple as possible. I think I'll keep it this way unless you can desribe an important use case where this is just bad behavior. The current use for this is just to get the vehicle off the ground so you can then fly manually.

lvale, 2017-04-26:
Let's just say that a powerful hexa/octo at the default take off height is extremely affected by ground effect...

DonLakeFlyer, 2017-04-26:
Hmm...

sparky002, 2017-05-01:
Likewise with powerful quads. Please don't remove features for simplicity sake. This is something that could cause a massive safety issue. At least put an option there to do so. That way it still remains simplified but adjustable for different users.

hamishwillee, 2017-05-02:
Maybe:
- UI element configurable based on type of vehicle or on an advanced mode selector in settings? 
- Link to param setting that configures takeoff height?

lvale, 2017-05-02:
@DonLakeFlyer @hamishwillee 

At least in ArduPilot, I think PILOT_TKOFF_ALT http://ardupilot.org/copter/docs/parameters.html#pilot-tkoff-alt-pilot-takeoff-altitude is not being used

hamishwillee, 2017-05-02:
I'm just updating the PX4 docs now. It uses [MIS_TAKEOFF_ALT](https://dev.px4.io/en/advanced/parameter_reference.html#MIS_TAKEOFF_ALT)

DonLakeFlyer, 2017-05-11:
Unfortunately PILOT_TKOFF_ALT defaults to 0!

Antiheavy, 2019-04-06:
I realize this is an old issue, but the current QGC documentation makes it look as if the request feature is now implemented?  https://docs.qgroundcontrol.com/en/FlyView/FlyView.html

DonLakeFlyer, 2019-04-19:
Yup it is

closed 2019-04-19 by DonLakeFlyer