mavlink/qgroundcontrol #5064: Vagrant using Qt 5.5

double16, 2017-04-26:
The README.md references QT 5.7 is required, but `.vagrantconfig.yml` references 5.5.

Looks like `.vagrantconfig.yml` needs to be updated.

dagar, 2017-04-26:
We don't use it in the build system, which is why it keeps getting left behind.
Do you use it @double16? Should we use it in the build system to keep it working?

double16, 2017-04-26:
Short answer: yes, please.

I'm new to QGC and I'd like to use Vagrant to avoid having to deal with keeping my dev machines up to date. I expect sporadically building QGC. I've experience pain on other projects in the past that involved having to diagnose dependency and tool issues. I would appreciate Vagrant being in the CI process so it keeps working.

I'm willing to giving a PR a shot if it's a matter of updating dependencies and making sure the tests run.

hamishwillee, 2017-05-17:
FYI, I just tried this in a build and this failed (lots of `==> default: cannot create symbolic link '/vagrant/shadow-build/release/Qt/libs/lib/libQt5Svg.so'` messages).

@dagar I was using this because it looked like the fasted way to get a daily build running to see how mocklink works. I imagine others who want to "just try stuff out" would like it too.

@double16 Are you planning on patching this?

```
==> default: echo Copying files && cp -f -R /vagrant/resources/flightgear /vagrant/shadow-build/release && mkdir -p /vagr
cc_64/lib/libQt5Core.so.5 /vagrant/shadow-build/release/Qt/libs/ && cp -f --dereference /tmp/Qt/5.5/gcc_64/lib/libQt5DBus
build/release/Qt/libs/ && cp -f --dereference /tmp/Qt/5.5/gcc_64/lib/libQt5Location.so.5 /vagrant/shadow-build/release/Qt
 --dereference /tmp/Qt/5.5/gcc_64/lib/libQt5MultimediaQuick_p.so.5 /vagrant/shadow-build/release/Qt/libs/ && cp -f --dere
5/gcc_64/lib/libQt5OpenGL.so.5 /vagrant/shadow-build/release/Qt/libs/ && cp -f --dereference /tmp/Qt/5.5/gcc_64/lib/libQt
so.5 /vagrant/shadow-build/release/Qt/libs/ && cp -f --dereference /tmp/Qt/5.5/gcc_64/lib/libQt5Qml.so.5 /vagrant/shadow-
s/ && cp -f --dereference /tmp/Qt/5.5/gcc_64/lib/libQt5QuickWidgets.so.5 /vagrant/shadow-build/release/Qt/libs/ && cp -f
tmp/Qt/5.5/gcc_64/lib/libQt5Sql.so.5 /vagrant/shadow-build/release/Qt/libs/ && cp -f --dereference /tmp/Qt/5.5/gcc_64/lib
ant/shadow-build/release/Qt/libs/ && cp -f --dereference /tmp/Qt/5.5/gcc_64/lib/libQt5Widgets.so.5 /vagrant/shadow-build/
 cp -f --dereference /tmp/Qt/5.5/gcc_64/lib/libicudata.so.56 /vagrant/shadow-build/release/Qt/libs/ && cp -f --dereferenc
/lib/libicuuc.so.56 /vagrant/shadow-build/release/Qt/libs/ && cp -f --dereference --recursive /tmp/Qt/5.5/gcc_64/plugins/
 /vagrant/shadow-build/release/Qt/plugins/ && cp -f --dereference --recursive /tmp/Qt/5.5/gcc_64/plugins/iconengines /vag
shadow-build/release/Qt/plugins/ && cp -f --dereference --recursive /tmp/Qt/5.5/gcc_64/plugins/platforminputcontexts /vag
dow-build/release/Qt/plugins/ && cp -f --dereference --recursive /tmp/Qt/5.5/gcc_64/plugins/position /vagrant/shadow-buil
se/Qt/plugins/ && cp -f --dereference --recursive /tmp/Qt/5.5/gcc_64/plugins/xcbglintegrations /vagrant/shadow-build/rele
ant/deploy/qgroundcontrol-start.sh /vagrant/shadow-build/release && cp -f /vagrant/deploy/qgroundcontrol.desktop /vagrant
==> default: Copying files
==> default: cp:
==> default: cannot stat '/tmp/Qt/5.5/gcc_64/lib/libicudata.so.56'
==> default: : No such file or directory
==> default: make:
==> default: *** [release/QGroundControl] Error 1
==> default: cp:
==> default: cannot create symbolic link '/vagrant/shadow-build/release/Qt/libs/lib/libQt5Location.so.5.5'
==> default: : Protocol error
==> default: cp:
==> default: cannot create symbolic link '/vagrant/shadow-build/release/Qt/libs/lib/libQt5Svg.so'
==> default: : Protocol error
==> default: cp:
```

dagar, 2017-05-17:
I've been dragging my feet because I don't really know how to use Vagrant yet.

If you list the commands to bring it up I'll add it to travis-ci in a pull request and we can pick away at the problems causing it to fail.

The majority of it is likely just updating Qt 5.5 to 5.7. https://github.com/mavlink/qgroundcontrol/blob/master/.vagrantconfig.yml#L3

double16, 2017-05-17:
I've done some work on it. I'm stuck now at pulling down the Qt 5.7 package. Qt 5.5 is on the project's S3 bucket and I'm not sure it's the same archive as the Linux package downloaded from Qt. I can start a pull request for what I've done so far.

@dagar The command is `vagrant up`, which creates the VM and builds the software. I'd be interested to see that work on Travis :)

dagar, 2017-05-17:
This is what travis does for Qt.
https://github.com/mavlink/qgroundcontrol/blob/master/.travis.yml#L92

The main problem is our travis-ci builds are already painfully slow. Do you happen to know how to do it with docker instead of a VM?

double16, 2017-05-17:
It's `vagrant up --provider docker` but there needs to be a base box that supports it. Should be one for Ubuntu 14.04, I'll look.

dagar, 2017-05-17:
If we simply had a docker image instead would that be preferred over Vagrant?

double16, 2017-05-17:
Docker isn't designed for UI, so it would be difficult, probably impractical, to construct an interactive (aka IDE) dev environment in docker.

double16, 2017-05-17:
I was able to get a Vagrant box on docker building to the same point as the VM. I need this to finish:
`https://s3-us-west-2.amazonaws.com/qgroundcontrol/dependencies/Qt5.7.1-linux.tar.bz2`

DonLakeFlyer, 2017-05-17:
I'm not sure about this vagrant stuff. Unless someone signs up to maintain it. It just ends up being too much stuff to keep track of.

double16, 2017-05-17:
The project README states Ubuntu is supported for native builds. There shouldn't be much difference between building on Ubuntu natively and in Vagrant. Not sure why there is a Qt package in this projects S3 bucket. Is that used for native Ubuntu builds? Maybe because Qt requires user intervention to install the developer packages.

hamishwillee, 2017-05-17:
> The project README states Ubuntu is supported for native builds. There shouldn't be much difference between building on Ubuntu natively and in Vagrant.

Absolutely. The dependencies are exactly the same. The usual process is for normal build process and vagrant to share a common dependency list (script or whatever) that Vagrant uses. 

The trick here is to understand where the rest of the system gets its dependencies and make sure that vagrant uses those.

DonLakeFlyer, 2017-05-17:
Maybe some way to link it to how we do travis then

double16, 2017-05-17:
I can look at it, it'll take some more time. Maybe over the next couple days.

hamishwillee, 2017-05-17:
That would be awesome. I'm a huge fan of vagrant as an consistent and reliable way for people to get started on a common build system. As @dagar says though, it has to be maintainable.

oliverengineer, 2018-08-03:
Hi everyone...
I managed to lunch qgc on vagrant ,but after that i don't know what i can do.
Can anyone please tell me how can i use qgc on vagrant?

saadIril, 2019-09-07:
@oliverengineer same here... Do we have to follow the native ubuntu build or qt and stuff is already installed?

DonLakeFlyer, 2019-09-07:
At this point I don't have time to maintain vagrant. Unless someone else picks it up I'm going to toss it from the build since it doesn't work.

DonLakeFlyer, 2019-12-24:
I may have fixed this. Will need someone who knows vagrant test test

PR #5169 (2017-06-02): Vagrant qt5.7 5064
Fixes #5064
closed 2019-12-24 by DonLakeFlyer