mavlink/qgroundcontrol #5082: Android: USB handling

AndKe, 2017-05-03:
If I reboot my phone, I get this:

[image]

I would expect this message when connecting an USB device, but it just spawns for no reason.

DonLakeFlyer, 2017-05-04:
@dogmaphobic I think the USB registration stuff for android is now pretty wide open to catch any usb device. This was on purpose to not have to chase a list of devices. This would seem to indicate there is some strange default usb device being registered on the phone right?

AndKe, 2017-05-04:
well, if the phone actually senesed any USB device, then at least Tower, maybe , 3DR services, Solo and Solex would spawn similar message. Or one of the thermal camera applications (that use other USB devices.)
All of them work as expected, and only QGC spawns this popup on reboot.

DonLakeFlyer, 2017-05-04:
If I remember correctly those apps have usb device registration for specific ids.

dogmaphobic, 2017-05-04:
Odd that the dialog does not tell you what USB device it is talking about. Is this peculiar to this one device or is this a general problem? I'd be curious if this phone has some internal device connected to the USB bus. Also strange that only QGC is offered as an option, which proves that it asking for anything while the other apps are a bit more discriminate. A logcat of the boot process would tell you what Android is finding and why it is asking you, but that implies you would have the proper dev tools installed.

DonLakeFlyer, 2017-05-06:
What kind of phone is this?

I'd rather not go back to being too specific on device connection since that causes all sorts of other problems.

AndKe, 2017-05-06:
well, that's a Note4 running a Note7 ROM  - so it's expected to behave the
same on newer Samsung devices.
I am not sure if it's reproducible on "older" (few) Samsung firmware on
Note 10.1 ,will check.

DonLakeFlyer, 2017-05-11:
Thanks @AndKe . I think for now we leave as is. Since specifying specific boards leads to a whole different set of problems which are worse. If we start seeing more reports of this on other device types then I think we will be stuck with going the more specific route. But for now I'd rather wait to see that first.

closed 2017-05-11 by DonLakeFlyer