mavlink/qgroundcontrol #5101: Windows application often takes several tries to launch

Antiheavy, 2017-05-06:
happens on many different PCs. sometimes takes many tries to launch application. load times is highly variable from a couple seconds to a minute.    Have noticed this on the daily builds for the past month or two.

DonLakeFlyer, 2017-05-06:
@NaterGator Do you think you could help out tracking this one down?

NaterGator, 2017-05-07:
Yup, @antiheavy any chance you can capture a crash minidump?

DonLakeFlyer, 2017-05-08:
Thanks Nate

NaterGator, 2017-05-08:
@Antiheavy I'm struggling to repro with the past few daily builds. Can you give some details about your system configuration, a crash minidump, a log trace of  startups, or anything that may help track this down?  You can use [DebugView++](https://debugviewpp.wordpress.com/) to externally capture the content of the QGC console.

Antiheavy, 2017-05-08:
I'm out of the office all day. I'll give it a try tonight. And reportback.

jonrwat, 2017-05-08:
I got a crash today when trying to run a branch (https://github.com/dogmaphobic/qgroundcontrol/tree/telemetryRSSI).  Here's the log files.
[QGC Error Logs.zip](https://github.com/mavlink/qgroundcontrol/files/984906/QGC.Error.Logs.zip)

jonrwat, 2017-05-08:
Additionally, I got the crash after building and attempting to run a "Release" build.  I'm currently not able to build a "Debug" build because I get dependency errors; not sure indirectly corresponds to the same issue or not.  Here's the dependency issue I get:

[image]

DonLakeFlyer, 2017-05-08:
> I'm currently not able to build a "Debug" build because I get dependency errors; not sure indirectly corresponds to the same issue or not. Here's the dependency issue I get:

Delete you build directory and rebuild again.

NaterGator, 2017-05-09:
@jonrwat the call stack from that minidump suggests the crash happened inside of the nvidia driver's rendering pipeline. 

Was that build created personally or downloaded from the build server? If it was personal do you mind trying the build server version from here: https://ci.appveyor.com/project/mavlink/qgroundcontrol/build/artifacts 

That will allow me to pull a more detailed stack trace and hopefully unwind what the application was doing around the time of the crash.

jonrwat, 2017-05-11:
@NaterGator 

>the call stack from that minidump suggests the crash happened inside of the nvidia driver's rendering pipeline.
I've historically had wonky behavior with by graphics card (and QGC and Qt specifically), so this issue may not be something to delve too far into (i.e. could be an issue with my PC).

> Was that build created personally or downloaded from the build server?
The build was created personally.  It was a "Release" build, using Qt Creator (Community) version 5.5.1, on the previously listed Git branch.

> If it was personal do you mind trying the build server version from here
I installed the build and opened QGC in Windows a dozen times without issue.  Maybe @Antiheavy can try on his PC to see if he gets similar results.

NaterGator, 2017-05-13:
@Antiheavy any chance you've got time to look at this again? 

> I installed the build and opened QGC in Windows a dozen times without issue. Maybe @Antiheavy can try on his PC to see if he gets similar results.

Right, I'm not blaming your build process (though 5.5.1 is older than rec'd), it is just that the build server stores the debugging symbols so I can unwind and inspect a stack trace. Without the PDB files the stack trace is almost completely devoid of useful info.

DonLakeFlyer, 2017-05-26:
@Antiheavy Can you reproduce using a standard daily build?

Antiheavy, 2017-05-26:
finally got back to this.

Recent mainline daily builds (windows) seem to install successfully and launch successfully every time.

I went back and installed some older builds from March/April and they show inconsistent launching.

I'm satisfied something has changed between a month ago and now.  Now seems better.  We will test on more and different laptops to be sure.

Antiheavy, 2017-05-27:
I did most testing of recent mainline daily on several other PCs.  Overall the load times for QGC are fairly variable.  The first time you launch it after fresh install or after reboot, it is slower to load (up to 2 minutes sometimes on old/slow PCs, as fast as 4 seconds on fast PCs).  Subsequent application launches typically launch ~2X faster, but the load times can be variable.   Older QGC builds (march/april) were slower to load overall compared to recent dailies so this effect is much more pronounced.

So:  recent daily builds seem to be working, although slow/variable to load on old/slow PCs.

@NaterGator is there anything else you'd like us to try here?  or are we calling this good enough?  we are still having problems with our custom QGC-branch successfully launching, are you willing to help us look into that branch too?

DonLakeFlyer, 2017-05-27:
@Antiheavy I should be able to track down what is going on with your custom branch. Just need to set up a little more build stuff to get the information needed. Was hoping to hit it here since things like symbols are already set up better in mainline.

NaterGator, 2017-06-01:
I suspect this initial load slowness is due to QML compilation and caching. Qt 5.9 (a new LTS release) purports to improve upon this (and allow us to frontload the compile during the build stage). 

@Antiheavy I am willing to look into your custom branch too presuming I can repro, but I need to do that during the weekends / on my own time in the evening if that is okay.

dogmaphobic, 2017-06-01:
QtQuick compiler has been around for a while. Currently, only the commercial licensed Qt has it. I've been using it for the iOS builds.

NaterGator, 2017-06-01:
@dogmaphobic yep, I was referring to this: http://blog.qt.io/blog/2017/05/31/performance-improvements-with-qt-5-9-lts/

DonLakeFlyer, 2017-06-20:
The crashing seems to have gone now for @Antiheavy so closing.

closed 2017-06-20 by DonLakeFlyer

Antiheavy, 2017-06-20:
yeah, depending on the day, the daily build will take longer or shorter to launch the first time.   For example recent daily builds are taking longer again (up to a minute?).  I've categorized this as a minor annoyance that we are living with for now.

DonLakeFlyer, 2017-06-20:
Need the Qt Compiler from commercial version.