mavlink/qgroundcontrol #5109: where did vehicle type go ?

AndKe, 2017-05-08:
when planning (not connected), I would expect to choose between plane & VTOLs right here:
[image]

before, the setting was in QGC configuration, now it's gone ?

DonLakeFlyer, 2017-05-08:
Click on Vehicle Info

AndKe, 2017-05-08:
By adding "survey" , the vehicle type and AP type is hidden (like the screenshot above). 
Please *disable* the vehicle type and AP type, instead of hiding it, it could even be useful when loading and inspecting a saved mission.

DonLakeFlyer, 2017-05-08:
Vehicle Info goes away as soon as you add a mission item. This is by design. A mission and the mission items themselves are targetted to firmware/vehicle type. This is because the behavior of the mission items values as well as what fields are actually available are specific to a firmware/vehicle combination. As soon as you add an item to the mission you have locked down the full set to the firmware/vehicle type.

AndKe, 2017-05-08:
Yes, I know it's by design, but if the design were to disable the droplists, and still let them be visible, there would not be any confusion. 
Also, should you load a mission a year later, or receive from somebody else, you would instantly know what it is made for.
So please consider not hiding, but instead disabling them from being changed.

DonLakeFlyer, 2017-05-08:
Ah ok, I see. I changed that somewhere along the way which was wrong.

closed 2017-05-18 by DonLakeFlyer