mavlink/qgroundcontrol #5112: Please add telemetry RSSI,  dBm / packet loss etc

AndKe, 2017-05-08:
RC RSSI has a prominent place, yet , that's barely used at all as it's only relevant during takeoff/landing,(where distance is short) while flight monitoring is using telemetry radios - and it's performance at great distance is much more interesting.

DonLakeFlyer, 2017-05-08:
@dogmaphobic Isn't telemetry RSSI already there?

AndKe, 2017-05-08:
as so often before, I am using tlog playback for testing, in this case AP3.7.1  with messages like:
```
2017-03-22 11:29:58.76: RADIO {rssi : 176, remrssi : 156, txbuf : 100, noise : 34, remnoise : 97, rxerrors : 0, fixed : 0}
2017-03-22 11:29:58.76: RADIO_STATUS {rssi : 176, remrssi : 156, txbuf : 100, noise : 34, remnoise : 97, rxerrors : 0, fixed : 0}
2017-03-22 11:29:59.68: RADIO {rssi : 168, remrssi : 155, txbuf : 100, noise : 37, remnoise : 94, rxerrors : 0, fixed : 0}
2017-03-22 11:29:59.68: RADIO_STATUS {rssi : 168, remrssi : 155, txbuf : 100, noise : 37, remnoise : 94, rxerrors : 0, fixed : 0}
2017-03-22 11:30:00.68: RADIO {rssi : 157, remrssi : 145, txbuf : 100, noise : 29, remnoise : 91, rxerrors : 0, fixed : 0}
2017-03-22 11:30:00.68: RADIO_STATUS {rssi : 157, remrssi : 145, txbuf : 100, noise : 29, remnoise : 91, rxerrors : 0, fixed : 0}
2017-03-22 11:30:01.82: RADIO {rssi : 167, remrssi : 146, txbuf : 100, noise : 37, remnoise : 92, rxerrors : 0, fixed : 0}
2017-03-22 11:30:01.82: RADIO_STATUS {rssi : 167, remrssi : 146, txbuf : 100, noise : 37, remnoise : 92, rxerrors : 0, fixed : 0}
2017-03-22 11:30:02.74: RADIO {rssi : 168, remrssi : 144, txbuf : 100, noise : 30, remnoise : 99, rxerrors : 0, fixed : 0}
2017-03-22 11:30:02.74: RADIO_STATUS {rssi : 168, remrssi : 144, txbuf : 100, noise : 30, remnoise : 99, rxerrors : 0, fixed : 0}
```

DonLakeFlyer, 2017-05-08:
@dogmaphobic Do these mavlink values sync with the changes here: https://github.com/mavlink/qgroundcontrol/pull/5110?

dogmaphobic, 2017-05-08:
I'm not sure I understand the question.

>RC RSSI has a prominent place, yet , that's barely used at all as it's only relevant during takeoff/landing

Does it mean RC RSSI is only relevant during takeoff/landing?

>while flight monitoring is using telemetry radios - and it's performance at great distance is much more interesting.

"it's" or "its"? It makes a difference :)

Either way, #5110 fixes a bunch of issues for Radio Telemetry, especially when using something other than the SiK radio. I would appreciate feedback on that while at it.

dogmaphobic, 2017-05-08:
Ha! I missed the "Add" part of the title as I got confused with the rest. It's always been there but it only shows if Radio Telemetry is present. Until before #5110, only the SiK radio would produce the proper values that would indicate it was present. I'm not even sure at this point if even that was working. As I said, #5110 fixes a lot of issues with Radio Telemetry and it should now show up if present.

AndKe, 2017-05-08:
RC RSSI on GCS, is never used by me, on near missions (under 2km) it's simply not an issue, not to mention Graupner's own telemetry would warn long before loosing connection.  On any bigger mission, the RC controller is usually off after entering auto mode anyway.

it is, and it's is very different, maybe I should not have written it's but but "their" (as in radios) - my point was that "their (radios) performance/status" was important..  English is not exactly my native language, I am better at two others  :)

dogmaphobic, 2017-05-12:
Has anybody tested after #5110 went in? Is it working?

AndKe, 2017-05-13:
I just tested HM-TRP connection, using FTDI adapter, so the VID,PID was not 0403:6015 as for 3DR radio on USB, but I doubt QGC checks for that.
I could not find Telemetry RSSI anywhere in current master GUI (RADIO and RADIO_STATUS mavlink packages were transmitted)

DonLakeFlyer, 2017-06-27:
@AndKe Anyway to get a telemetry log file for this test case? You shouldn't need to actually fly, just connect up I think. As long as RADIO_STATUS is getting through.

AndKe, 2017-06-27:
@DonLakeFlyer  
[2017-06-23 12-58-30.tlog.zip](https://github.com/mavlink/qgroundcontrol/files/1105036/2017-06-23.12-58-30.tlog.zip)

DonLakeFlyer, 2017-06-28:
@dogmaphobic Can you take a look at this? I tried playing back this telemetry log and the RADIO_STATUS messages don't seem to come through. I then tried just to connect up to a board via SiK Radio and I don't see RADIO_STATUS messages coming through that way either. Not sure what is going one here.

DonLakeFlyer, 2017-06-29:
@dogmaphobic I figured this out. At some point you moved handling of RADIO_STATUS from MAVLinkProtocol to Vehicle. Although Vehicle does receive all mavlink messages it filters out all messages which do not come from the vehicle based on system id comparison. Since RADIO_STATUS doesn't come from the vehicle, it comes from the radio and has it's own system id which does not match the vehicle id. This means that all RADIO_STATUS messages are dropped on the floor by Vehicle. I'm trying to figure out how to rearrange things to make it work from Vehicle.

DonLakeFlyer, 2017-06-29:
@dogmaphobic Pull #5359 fixes the problem of Vehicle never seeing RADIO_STATUS messages. But it still does not work (No Indicator). I'll leave the rest for you to look at.

dogmaphobic, 2017-06-29:
I run into a similar scenario the other day. That made me think about situations like this. My conclusion was that the _system_ ID should be the same. Ultimately, it's all one system. Only the _component_ ID should be different.

DonLakeFlyer, 2017-06-29:
> My conclusion was that the system ID should be the same. Ultimately, it's all one system. Only the component ID should be different.

That's not possible. A single radio can have multiple vehicles passing through it. With each vehicle having a different system id.

DonLakeFlyer, 2017-06-29:
I was wrong. This is working now. If you replay the attached log the indicator shows up. So I think this is all working now.

closed 2017-06-29 by DonLakeFlyer