mavlink/qgroundcontrol #5132: Top portion of video window is obscured by toolbar

jaxxzer, 2017-05-11:
The toolbar sits on top of the video and map. I think the top of the video should be anchored to the bottom of the toolbar, so everything in the video is clearly visible.

jaxxzer, 2017-05-11:
If we want to do this, it's an easy fix: https://github.com/bluerobotics/qgroundcontrol/commit/e11f072b7ccca7e43115a0b808bef3230d0d8283

DonLakeFlyer, 2017-05-11:
@dogmaphobic ?

dogmaphobic, 2017-05-12:
It depends on the screen and the aspect ratio. By making it anchor on the bottom of the toolbar, it will make the video window shorter and inherently narrower because of the aspect ratio. This would work fine on desktop as you can size the window any way you want but it will impact mobile devices with fixed screen sizes. In the end, it's a matter of taste. Do you want black bars on the sides or not. I would rather fill the screen and crop the image. It could easily be made into an option. The question then is where to add the option.

closed 2026-10-05 by github-actions[bot]