mavlink/qgroundcontrol #5136: Autotakeoff problems.

AndKe, 2017-05-12:
Please see attached tlog.
I did a simple test: (using QGC master:da811e1a8 from 2017-05-05 14:08:38)
Created a mission for connected copter (AC 3.4.4 , APM stack, on Pixhawk)
Mission items were takeoff (to 50m) , then a few waypoints, and LAND.
QGC had a "slide-to-takeoff" or similar item, I used it , but got an error that "Guided mode not supported by hardware" (or similar wording/meaning)

So I tried the left pane "takeoff" button , it worked, switched to guided, took off to 2.5m .. and descended after a while, needless to say, needed manual assertion of AUTO to actually run the mission.

1: why did not the "run mission (start with autotakeoff work)
2: it complained by error, that guided mode were not available.
3: the .tlog attached cannot be played back using [file][replay flight data]

[QGC-noAutoTakeoff.tlog.zip](https://github.com/mavlink/qgroundcontrol/files/997323/QGC-noAutoTakeoff.tlog.zip)

DonLakeFlyer, 2017-05-12:
This isn't working yet on daily builds with ArduPilot. There are already separate issues in for each problem already. Should be fixed in a week or so.

If the vehicle descended without you telling it or doing anything from QGC side then that would be a firmware problem. Could it have run out of battery and been doing an RTL?

closed 2017-05-12 by DonLakeFlyer

AndKe, 2017-05-12:
Thanks .
as for the descend,  it was not a power issue, I have very solid battery monitoring (with madhacker Mav2HoTT), and it flew fine after switching to Auto (it's not exactly a new UAV) - anyway - I agree that it would not be caused by guided unless commanded in guided mode, which I have no reason to believe QGC did.
- so sorry for a duplicate issue :)

DonLakeFlyer, 2017-05-12:
> so sorry for a duplicate issue :)

Don't worry about at that at all. I'd rather have dups than miss something.