mavlink/qgroundcontrol #5142: Autodetect other radios, and auto connect behavior.

AndKe, 2017-05-13:
QGC, as is, detects FTDI (VID:PID 0403:6015) as 3DR radio , and auto-connects just fine. - everything else requires several clicks to connect manually.

Is it ok to add same behaviour for 0403:6001 , that's another FT323 product used in FTDI cables, I use them for better S/N ratio than when providing USB to the radio.

Finally, about same goes for 10c4:ea60 (yet another radio on CP210) also 57600baud, could very well be auto-connected. 

Any thoughts ?

AndKe, 2017-05-13:
please see https://github.com/mavlink/qgroundcontrol/pull/5143
both tested and found good. - CP210 device was apparently attempted supported before, bug corrected, and FTDI cable works fine as well.

DonLakeFlyer, 2017-05-15:
New radios are easy. Different baud rate is not.

closed 2026-10-05 by github-actions[bot]