mavlink/qgroundcontrol #5161: Windows daily builds are not getting pushed to S3

DonLakeFlyer, 2017-05-16:
@dagar Can you take a look?

dagar, 2017-05-19:
This is weird, according to many appveyor builds it's still successfully deploying QGroundControl-Installer.exe

dagar, 2017-05-19:
Made a couple changes at the appveyor settings level. Test build https://ci.appveyor.com/project/mavlink/qgroundcontrol/build/master-5035

DonLakeFlyer, 2017-05-21:
Still not working as far as I can tell

tubeme, 2017-05-22:
Not working yet...

dagar, 2017-05-25:
I'm really at a loss here. Nothing has changed in the appveyor config since it stopped working.
Trying again https://ci.appveyor.com/project/mavlink/qgroundcontrol

tubeme, 2017-05-25:
I get this error right now:

<Error>
<Code>AccessDenied</Code>
<Message>Access Denied</Message>
<RequestId>B1A7F1D762F084D3</RequestId>
<HostId>
GW2W8HIG7olBZ0KhLqCxImwlxNA1paq4PpliBUvmw3klUKDuEdojhnkYC6/B46bNZo96vpWdPes=
</HostId>
</Error>

DonLakeFlyer, 2017-05-26:
Fixed

closed 2017-05-26 by DonLakeFlyer