mavlink/qgroundcontrol #5163: Intrument panel overlaps multi-vehicle radios

DonLakeFlyer, 2017-05-16:
[image]

DonLakeFlyer, 2017-05-16:
@dogmaphobic Can you take this one. The code here seems different from last I looked. Things used to be top anchored to the bottom of these radios. Now the instrument stuff seemds to be anchored to the bottom. Don't want to screw anything up here with plugin and so forth.

dogmaphobic, 2017-05-16:
Gazebo is broken. That's the only way I know how to test multiple vehicles (other than actually starting multiple vehicles, which is a bit of a pain :P ) I will take a look when that gets fixed.

DonLakeFlyer, 2017-05-16:
Start up two MockLink vehicles

closed 2017-06-22 by DonLakeFlyer

hamishwillee, 2017-06-22:
And just should anyone find this in future, there is information about [Mock Link here](https://dev.qgroundcontrol.com/en/tools/mock_link.html)