mavlink/qgroundcontrol #5182: Proposal: Timers for professional use.

tubeme, 2017-05-22:
Hello guys,

I might be wrong but I could not find a way to follow my timing in the instrument panel.

In real airplane there are 3 clocks Local time, UTC and flight time.

It will be very useful if we have this information somewhere in the instrument panel: Local time; UTC; Flight Time (since arming).

It is mandatory for commercial use.

If there is such info please give me pointers?

thanks

DonLakeFlyer, 2017-05-22:
Flight time is available from instrument panel in daily builds. New users automatically get it with default setup. Existing users who have already modified instrument panel values need to add manually. Is current time really needed?

tubeme, 2017-05-25:
I saw it is avail from an image you posted. But I really cannot download the daily's and my latest master is outdated and it is without the timers...

For a pro work, and not a pro work for geodesy, there are teams working in some scenarios. Especially involving teams syncing. Either if we fly a couple of machines at a couple of locations or if we sync our observation work with other team members and with a ground observation. 

In order for this sync to happen we usually do a very strict and detailed flight plans, which involve timing.

So for convenience it is very good to have this timers at our disposal in the ground station. Not that we don't do it with the timer of the computer we install the GC on, but it is distracting for the pilot this way.

The UTC time is very good if we have to sync work across time zones. If it is not that hard to implement we would love to have this timers. If not we will do it the classical way. But this is part of the user experience.

DonLakeFlyer, 2017-12-30:
@jaxxzer Didn't you just add something like this?

jaxxzer, 2018-01-03:
I added a wall clock readout to the vehicle facts/values widget. I think the request is for a countdown timer which would probably need to go in a new widget.

booo, 2022-10-31:
Implemented in 123e64e9107a05ba30c40c00d816a53b108fc689

closed 2022-10-31 by booo