mavlink/qgroundcontrol #5193: Survey - Spacing Parameter

tops4u, 2017-05-24:
Currently I'm flying Search Missions to find fawn in Farmlands. My FLIR Cam will cover about 25m with some added on safety margin I'll cut it down to 20m as the area was also uneven. So I did a survey and entered 20m as Grid Spacing. Now I noticed that at the Beginning / Ending sometimes the Survey would not fill up completely. One might argue that this is the nature of this parameter - if this is the case please add some more description on the '?' - Marks explanation. 

My expectation was: 
- I draw the Search - Area
- Enter a Spacing - which corresponds to my cameras view angle
- QGC would fill up so that the whole Area is covered. 

As can be seen on the Image below, that the last bit of the survey does not completely round up to the 20m Spacing so the last bit of around 10m broad would be left out. This is the End Part:

[image]

Vs the Measurement in Google Maps:

[image]

In this example this is peanuts as there is a house, in other scenarios it might be hard to tell if 100% of the Field is searched or not. This is important for this kind of mission. 

On the other hand on the Beginning Side there would have been some space left... 

[image]

[image]

DonLakeFlyer, 2017-05-24:
I'm going to need to .plan files for these to reproduce.

tops4u, 2017-05-25:
Here is the file. 

[Fröschern-Aetingen.plan.zip](https://github.com/mavlink/qgroundcontrol/files/1028404/Froschern-Aetingen.plan.zip)

DonLakeFlyer, 2017-05-25:
There is also a problem in latest daily builds where grid angle is reversed.

DonLakeFlyer, 2017-05-25:
@antiheavy Curious if you have experience with other product which solve this problem. The issue is when the transect it just outside the bounds of the survey polygon. It will get clipped out. Which will mean you will not end up getting images over part of the area. I tried both Mission Planner and DroneDeploy and they both have exactly the same problem.

Here is a debug image which shows the transects prior to clipping. It shows an extreme case where you lose both left and right edges of the polygon.

[image]

DonLakeFlyer, 2017-05-25:
@WickedShell Do you deal with this?

WickedShell, 2017-05-25:
@DonLakeFlyer Nope, given that for good photogrammetry you want extra coverage outside the target area, I've ignored this edge case as not being a common one. (and just extend the coverage polygon slightly if its a large concern. You could try and create extra lines around the outside edges, but it gets really complicated/unexpected quickly.

tops4u, 2017-05-25:
I could live with that, the problem for the 'normal' User is, that this behaviour is not obvious or at least I did not expect it to work this way. 

My Expectation was, that the whole area inside the Polygon would be covered and the Distance (Spacing) between the Flying-Paths would be equal to my Camera coverage. As it is now my Coverage will miss out parts at the edges and this is not visible for me.

WickedShell, 2017-05-25:
@tops4u the problem you quickly run into is that (again speaking from a photomapping perspective) a single outside line/edge isn't enough coverage for processing. The beast automated response is to grow the survey region by an oversize parameter. (Requires a bit of math for none awful oversizing, implementation wise shouldn't be to bad, you just have to walk your polygon and push out against the winding direction, I just didn't rate it that high as a priority)

DonLakeFlyer, 2017-05-25:
> You could try and create extra lines around the outside edges, but it gets really complicated/unexpected quickly.

Yeah I can see that. I'm going to move to 3.3. But even then I'm not confident there is any sane way to fix this problem.

Antiheavy, 2017-05-25:
Ha!  yep, this is a common problem in the various tools I've used.

I've seen some tools that will try to center the flight paths (transects) inside the polygon such that the space on the two sides is equal and minimized.  i.e. if the transect spacing is 20 meters then there will be a maximum of 10 meters between the sides of the polygon and the first/last transect. 

I've also seen tools that will cheat the transect spacing say +/- 10% or something to make things fit better, I don't like this approach as it screws with your overlap.

The polygon isn't really the area on the ground that will be imaged; it is the area for the aircraft to fly in.  Calculating flight paths based on estimate of camera pixels on the ground would be interesting, but nobody really does that because it is a can of worms.

WickedShell's suggestion of drawing the polygon slightly large so that your flight paths cover what you need is the simplest approach.

tops4u, 2017-05-26:
Sorry for my stupid question, but why not doing it like this? Add one more Transection and place everything in the middle of the polygon? Or alternatively add one more that has a narrower spacing? 

Currently it might be hard to tell if you leftout some part. If you draw it bigger there might be a problem that there are obstacles you want to avoid (Trees for instance) or that the Camera will look at much warmer things (parked Cars, Streets) which will influence the AGC so that fawn may not be found reliably. This is not a Problem at turnaround points but when the transection is at the edge. 

Sorry if I this gets too complicated.

DonLakeFlyer, 2017-05-26:
I'm sure there are lots of different way to solve this. But at this point none of them are going to make it into 3.2 Stable.

tops4u, 2017-05-26:
@DonLakeFlyer ok, I'll be patient ;-)

Some thing else I wanted to ask... Where did the Daily Build Download Page go to? On qgroundcontrol.org there are only the releases the Beta Page vanished. I don't have my own Build System and was happy using latest master builds. Sorry for mixing this thread up, but I didn't find other means to contact you.

DonLakeFlyer, 2017-05-26:
Daily builds are under Releases: https://docs.qgroundcontrol.com/en/releases/daily_builds.html.

DonLakeFlyer, 2019-12-11:
Lost in the fray of too many issues...

closed 2026-10-05 by github-actions[bot]