mavlink/qgroundcontrol #5212: ArduPilot: Look into flow support

DonLakeFlyer, 2017-05-26:


proficnc, 2017-05-27:
Thanks Don

DonLakeFlyer, 2018-05-25:
@proficnc What's the URL for the ArduPilot flow firmware?

proficnc, 2018-05-25:
http://download.ardupilot.org/downloads/wiki/advanced_user_tools/px4flow-klt-06Dec2014.zip

proficnc, 2018-05-25:
Thanks for looking at this

DonLakeFlyer, 2018-05-26:
How it will work. QGC remembers the last firmware type you flashed and uses that to prime the firmware type selection for the PX4 Flow as well:
[image]

So in a lot of cases this should by default end up matching what you want.

DonLakeFlyer, 2018-05-26:
@proficnc I need a url to .px4 file for the PX4 Flow. A zip file won't work.

DonLakeFlyer, 2018-05-26:
I'll ask the dev folks.

DonLakeFlyer, 2018-05-26:
This is all done and ready to go. I just need a url to download from.

proficnc, 2018-05-26:
I will ask @Tridge to put it somewhere better.

proficnc, 2018-05-27:
Thanks to Tridge...

http://firmware.ardupilot.org/Tools/PX4Flow/

closed 2018-05-28 by DonLakeFlyer