mavlink/qgroundcontrol #5215: Survey grid still go out of bounds

lvale, 2017-05-27:
When planning a survey grid where there are angles on the survey area, QGC does not plan for points within the survey area. Screenshot represents that, where the vehicle would fly above houses, and this is a issue with privacy
[image]

lvale, 2017-05-27:
Also with different survey angle, note a NaN on the Survey Area

[image]

DonLakeFlyer, 2017-05-27:
The first image is on purpose. Survey does not yet support concave polygons (or is that convex). Anyway it's the one you drew!

The last one is a bug.

closed 2026-10-05 by github-actions[bot]