mavlink/qgroundcontrol #5222: Some Functions are not working with Arducopter 3.2.1 on an APM

Idefix0496, 2017-05-28:
It shows me always -1 mAh for the Already Consumed mAh
I can't set Rally Points
There are some Pop-Ups which say that some Parameters are missing, therefore one thing which is restricted is the Auto_Takeoff

DonLakeFlyer, 2017-05-29:
Can you tell me which parameters is says are missing?

Idefix0496, 2017-05-29:
[image]
Hi
I've attached the Errors I could reproduce. The Upload of Missions is sometimes working.

DonLakeFlyer, 2017-06-04:
I took a look at this. 3.2 is just too old with too many differences. The best QGC can support for ArduCopter is 3.3.

closed 2017-06-04 by DonLakeFlyer

hamishwillee, 2017-06-04:
Hi Don, 

how about  [here](https://docs.qgroundcontrol.com/en/) we change:
> Full setup/configuration of ArduPilot and PX4 Pro powered vehicles.
to
>Full setup/configuration of ArduPilot (3.3.3+) and PX4 Pro powered vehicles.

If reasonable, we could also publish the last QGC version that supports ArduCopter 3.3?