mavlink/qgroundcontrol #5231: Conversion from Knots to m/s on QGCXPlaneLink.cc is wrong

uthynauta, 2017-06-01:
I think the conversion factor on QGCXPlaneLink.cc in **master** (lines 638-640) is wrong:
      ind_airspeed = p.f[5] * 0.44704f;
      true_airspeed = p.f[6] * 0.44704f;
      groundspeed = p.f[7] * 0.44704;

to convert from knots (NM/hr) to m/s, the correct one should be 0.514444.
There are two types of miles, nautical and statute. In aeronautics and naval terms NM are used always. Therefore, the issue is that 1 kt = 1 NauticalMile/hr, **NOT** 1 StatuteMile/hr. 

Code is here: https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/QGCXPlaneLink.cc

I'm just learning GIT so I'm struggling to request this pull, I don't know if anyone can help me with this.

DonLakeFlyer, 2017-06-09:
How did you determine there was something wrong here and the value from X-Plane is actually nautical miles? Is there some doc on what units come across from X-Plane?

uthynauta, 2017-06-09:
Don,
When sending packets from X-PLane via UDP a box appears on the upper-left corner displaying the data that's being send through the UDP port, these data shows that all speeds (IAS, TAS and CAS) are sent in knots which are NM/hr. I'm not sure if there is an actual document where this is explicitely stated.

[image]

DonLakeFlyer, 2017-06-09:
Perfect, thanks. I'll fix it.

uthynauta, 2017-06-09:
Thank you very much Don

DonLakeFlyer, 2017-06-20:
@antiheavy Were you seeing wrong speed values when you did all your HIL testing?

Antiheavy, 2017-06-20:
huh.  to be honest I wasn't paying close attention to airspeed correctness in the HILstar model - i just sort of let it do it's thing.  Let me know if you guys would like me to test or verify anything tomorrow.

closed 2017-06-20 by DonLakeFlyer