mavlink/qgroundcontrol #5232: RTK GPS UI work and enhancements

mhkabir, 2017-06-02:
@LorenzMeier @dogmaphobic @DonLakeFlyer We need some UI work to get RTK GPS to a stage where it can be used by the average user. This is applicable for both APM and PX4 flight stacks. This would allow us to cleanly support the inexpensive UBlox M8P modules (e.g ProfiCNC Here+ or the Drotek XXL RTK). I have only tested with the Drotek module till now, and everything works internally. However, the status of Survey-in, whether RTK GPS is connected at all, etc. is only visible in the console and this is not suitable for an user.

### RTK status indicator
There is need for an icon on the top panel visible when RTK GPS connected - but _irrespective_ of whether a  vehicle is connected.

[image]

It would be an icon on the top bar (to the left of "Waiting For Vehicle Connection" maybe?) like the one we have for the vehicle's GPS status, and show the following information when clicked : 
- Satellite count for RTK base
- Accuracy of base position
- Survey-in status for base (survey-in position fixed or not, etc.)
- Time survey-in has been running if not completed.

### Map indication

The base station location derived from the RTK base station should be used and displayed as QGC's location on non-mobile platforms (the "Q" icon on the map).

### Software enhancements

#### Galileo support
The PX4 GpsDrivers stack needs update to support the RTCM corrections (message XXXX) from satellites in the Galileo constellation. I will send in a PR with these changes soon. On the M8P modules, firmware version 3.01 from UBlox is needed. 

#### MAVLink 2.0 support
We dearly need MAVLink 2.0 to optimize packet size for the RTCM injection messages. Currently we just send the full 182-byte message even though only the top 22 bytes or so are filled for some messages (e.g base position). MAVLink 2 will automatically chop off the empty packets - allowing for a much leaner transmission requirement for the uplink.

#### Verify bandwidth requirements
We will aim to support 3DR radios as the default telemetry link for RTK (since wifi has proved to be unreliable and short-range, even though it is high bandwidth.) With MAVLink 1 (no packet chopping), the datarate is too much and RTCM injection rate gets reduced. (this is for 3 RTCM messages @ 1Hz each). With the Galileo message, this link saturation will become even more of an issue. So we need to move to MAVLink 2, and as a last-resort, reduce the base position stream rate (to e.g 0.2Hz from 1 Hz) since it is always fixed.

Side note : I have started a PX4 Devguide entry for RTK here : https://dev.px4.io/en/qgc/rtk-gps.html

proficnc, 2017-06-04:
Great job documenting this! Getting RTK to be simple for all is critical!

DonLakeFlyer, 2017-06-04:
Yeah I had looked at doing a bunch of this work in relation to my work with Sentera since we were testing RTK support there. And also added support for Survey in accuracy and Minimum observation time in settings with that work.

That said I wonder if Survey support in 3.2 is really incomplete without at a minimum the RTK status indicator and Map indication work? What about the additional driver support? Opinions? @antiheavy FYI

DonLakeFlyer, 2017-06-04:
Also thanks @mhkabir for writing this all up.

mhkabir, 2017-06-05:
@DonLakeFlyer thanks! 

> And also added support for Survey in accuracy and Minimum observation time in settings with that work.

This is great. When is this making it upstream? :)

> That said I wonder if Survey support in 3.2 is really incomplete without at a minimum the RTK status indicator and Map indication work? 

I would agree. RTK is unusable from a user perspective without atleast some basic UI support. 

> What about the additional driver support?

I can add the additional driver support (BeiDou and Galileo corrections) by end of the day, so you're covered on that side. However, we must make sure that MAVLink 2.0 
 (https://github.com/PX4/Firmware/pull/7344) works properly, because of the bandwidth requirements. I wrote up a detailed section on bandwidth requirements for RTK (see "Uplink Datarate" section) here : https://dev.px4.io/en/tutorials/rtk-gps.html

DonLakeFlyer, 2017-06-05:
Does ProficCNC Here+ need new drivers in order to make it work?

> And also added support for Survey in accuracy and Minimum observation time in settings with that work.

This has been in upstream for a while now,

mhkabir, 2017-06-05:
@DonLakeFlyer No it should work on master without anything else - however, we don't use all the satellite information that the module has available. We are only using GPS and GLONASS constellations. BeiDou and Galileo are also supported by any M8P module, but we don't support those yet on the QGC side - and thus it needs a driver update.

DonLakeFlyer, 2017-06-05:
I think that part can wait for 3.3 then.

mhkabir, 2017-06-05:
OK, fine with me.

Antiheavy, 2017-06-09:
We've been playing around with the uBlox M8P on QGC and PX4 for a while now.  The current feedback is enough for some users, but one thing that would be helpful for the general user would be a status/progress indicator for the "Survey-In" function on the base station.

Currently we use the console output as a method to view Survey-In status, it's ugly, but it serves it's purpose for testing.
[image]
Ideally this would be replaced by some status and progress indicator that would automatically show up in the main Fly screen when QGC was connected to a uBlox M8P RTK basestation.

The feedback via PX4 for the aircraft seems acceptable to me as-is, but I suppose things could always be made fancier - I don't have any specific suggestions for improvement here:
[image]

DonLakeFlyer, 2017-06-22:
Stuff left here is support for additional driver support

closed 2019-01-19 by DonLakeFlyer