mavlink/qgroundcontrol #5238: It's so strange about UDP for Video Stream

HTaeHyeon, 2017-06-04:
[image]
[image]

I have Some Problems about Video Stream.
My computer's System is Windows.
And I use Gstream for windows,  but I can't use Video Stream for UDP.
Display always display the picture which attached image.
So I can't display my video. How do I do?

jaxxzer, 2017-06-08:
When launching in 'GPU compatability mode' on windows machine, I see the ArduSub image in the video window, sort of like this. The ArduSub image also appears red when this happens.

HTaeHyeon, 2017-06-09:
Thank you so much.
I got a problem. I've forgotten that I didn't install Graphic Driver.
After installing Graphic Driver, and then It works very well. Thank you.

closed 2017-06-09 by HTaeHyeon