mavlink/qgroundcontrol #5240: OSX daily drop is still broken due to missing QtLocation dylib

DonLakeFlyer, 2017-06-04:
@jaxxzer Has a repro of this on his machine.

tstellanova, 2017-06-05:
Verified nightly build crashing on my machine (see below).  Stable download works fine. 

Process:               QGroundControl [939]
Path:                  /Applications/QGroundControl-nightly.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               3.1.2 (1314)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QGroundControl [939]
User ID:               501

Date/Time:             2017-06-05 14:11:31.887 -0700
OS Version:            Mac OS X 10.12.5 (16F73)
Report Version:        12
Anonymous UUID:        1EEA34BA-B5C0-1AA4-F4FA-FDF7A873B798

Sleep/Wake UUID:       D33FBAD7-01B7-4B44-9BD1-D5C7D0D0BEB5

Time Awake Since Boot: 4400 seconds
Time Since Wake:       4300 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: @rpath/QtLocation.framework/Versions/5/QtLocation
  Referenced from: /Applications/QGroundControl-nightly.app/Contents/MacOS/QGroundControl
  Reason: image not found

Binary Images:
       0x1033fe000 -        0x103d37ff7 +org.qgroundcontrol.qgroundcontrol (3.1.2 - 1314) <0C0E814F-628D-34D5-81D3-BF6E55A36CB3> /Applications/QGroundControl-nightly.app/Contents/MacOS/QGroundControl
       0x103ea0000 -        0x103f36ff7  com.apple.framework.IOKit (2.0.2 - 1324.60.3) <7CE4C98B-107C-3AAA-B49A-F2ACFCBBF526> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
       0x103f7c000 -        0x104054fff +org.libsdl.SDL2 (2.0.4 - 2.0.4) <9A6817E0-DA9E-3EF6-9B14-1DB1367DCF67> /Applications/QGroundControl-nightly.app/Contents/Frameworks/SDL2.framework/Versions/A/SDL2
       0x104079000 -        0x104079fff +GStreamer (0) <2EE72D88-2525-38E1-AFAA-5755B58B5E3A> /Applications/QGroundControl-nightly.app/Contents/Frameworks/GStreamer.framework/Versions/1.0/lib/GStreamer
       0x109709000 -        0x109746dc7  dyld (433.5) <322C06B7-8878-311D-888C-C8FD2CA96FF3> /usr/lib/dyld
    0x7fffb7556000 -     0x7fffb7556fff  com.apple.ApplicationServices (48 - 48) <4C71CBA8-47E4-38BF-BE3B-F20DF8667D5D> /System/Library/Frameworks/ApplicationServices.framework/Versions/A/ApplicationServices
    0x7fffb8c30000 -     0x7fffb90c9ff7  com.apple.CoreFoundation (6.9 - 1349.8) <09ED473E-5DE8-307F-B55C-16F6419236D5> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation

Model: MacBookPro13,3, BootROM MBP133.0226.B23, 4 processors, Intel Core i7, 2.9 GHz, 16 GB, SMC 2.38f7
Graphics: Radeon Pro 460, AMD Radeon Pro 460, PCIe, 4096 MB
Graphics: Intel HD Graphics 530, Intel HD Graphics 530, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x15A), Broadcom BCM43xx 1.0 (7.21.171.126.1a2)
Bluetooth: Version 5.0.4f18, 3 services, 27 devices, 1 incoming serial ports
Network Service: USB 10/100 LAN, Ethernet, en7
Network Service: Wi-Fi, AirPort, en0
USB Device: USB 3.0 Bus
USB Device: iBridge
USB Device: USB2.0 Hub
USB Device: USB 2.0 Hub
USB Device: USB2.0 Hub
USB Device: PX4ESC
USB Device: FT230X Basic UART
USB Device: USB2.0 Hub
USB Device: CP2102 USB to UART Bridge Controller
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
USB Device: USB3.0 Hub
Thunderbolt Bus: MacBook Pro, Apple Inc., 19.6
Thunderbolt Bus: MacBook Pro, Apple Inc., 19.6

dogmaphobic, 2017-06-05:
The part of relevance in that crash report is this:

```
Dyld Error Message:
Library not loaded: .framework/Versions/5/QtLocation
Referenced from: /Applications/QGroundControl-nightly.app/Contents/MacOS/QGroundControl
Reason: image not found
```

It seems it's looking for things under `.framework` instead of `QtLocation.framework`

Could you please verify the contents of the Frameworks directory?

```
$ ls -l QGroundControl.app/Contents/Frameworks/
total 320
drwxr-xr-x  3 gus  staff     272 Dec  8 16:34 GStreamer.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtConcurrent.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtCore.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtGui.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtLocation.framework   <====== Is this one there?
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtMultimedia.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtMultimediaQuick_p.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtMultimediaWidgets.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtNetwork.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtOpenGL.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtPositioning.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtPrintSupport.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtQml.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtQuick.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtQuickControls2.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtQuickTemplates2.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtQuickWidgets.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtSerialPort.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtSql.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtSvg.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtTest.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtWidgets.framework
drwxr-xr-x  3 gus  staff     170 Jun  5 19:05 QtXml.framework
drwxr-xr-x  3 gus  staff     204 Feb 21 13:16 SDL2.framework
-rwxr-xr-x  1 gus  staff  160320 Jun  5 19:05 libbz2.1.0.dylib
```

tstellanova, 2017-06-08:
For the nightly build:
```
bromst:/ todd$ ls -l Applications/QGroundControl-nightly.app/Contents/Frameworks/
total 320
drwxr-xr-x@ 8 todd  admin     272 Jun  5 12:21 GStreamer.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtConcurrent.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtCore.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtGui.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtLocation.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtMultimedia.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtMultimediaQuick_p.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtMultimediaWidgets.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtNetwork.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtOpenGL.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtPositioning.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtPrintSupport.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtQml.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtQuick.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtQuickControls2.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtQuickTemplates2.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtQuickWidgets.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtSerialPort.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtSql.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtSvg.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtTest.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtWidgets.framework
drwxr-xr-x@ 5 todd  admin     170 Jun  5 12:38 QtXml.framework
drwxr-xr-x@ 6 todd  admin     204 Jun  5 12:04 SDL2.framework
-rwxr-xr-x@ 1 todd  admin  160496 Jun  5 12:38 libbz2.1.0.dylib
```
For the stable build (that doesn't crash, for me):
```
bromst:/ todd$ ls -l Applications/QGroundControl-stable.app/Contents/Frameworks/
total 320
drwxr-xr-x@ 8 todd  admin     272 Feb 14 20:39 GStreamer.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtConcurrent.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtCore.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtDBus.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtGui.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtLocation.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtMultimedia.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtMultimediaQuick_p.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtMultimediaWidgets.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtNetwork.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtOpenGL.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtPositioning.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtPrintSupport.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtQml.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtQuick.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtQuickWidgets.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtSerialPort.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtSql.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtSvg.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtTest.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtWidgets.framework
drwxr-xr-x@ 5 todd  admin     170 Feb 14 20:55 QtXml.framework
drwxr-xr-x@ 6 todd  admin     204 Feb 14 20:37 SDL2.framework
-rwxr-xr-x@ 1 todd  admin  160496 Feb 14 20:55 libbz2.1.0.dylib
```

DonLakeFlyer, 2017-06-27:
Fixed I hope!

closed 2017-06-27 by DonLakeFlyer