mavlink/qgroundcontrol #5245: andriod build error :arm-linux-androideabi-g++: error: oj\moc_SerialLink.obj: No such file or directory

sadmzyy, 2017-06-05:
[image]

# this is the configure of  qt, wnen I build qgc ,get this error as below:
mingw32-make: Nothing to be done for 'qmake_all'.
16:33:29: 进程"C:\Qt\Qt5.7.1\Tools\mingw530_32\bin\mingw32-make.exe"正常退出。
16:33:29: 正在启动 "C:\Qt\Qt5.7.1\Tools\mingw530_32\bin\mingw32-make.exe" "INSTALL_ROOT=E:\build-qgroundcontrol-Android_for_armeabi_v7a_GCC_4_9_Qt_5_7_1-Debug\android-build" install

linking debug\libQGroundControl.so
**_arm-linux-androideabi-g++: error: oj\moc_SerialLink.obj: No such file or directory_**
Makefile:1242: recipe for target 'debug\libQGroundControl.so' failed
mingw32-make: *** [debug\libQGroundControl.so] Error 1
16:33:33: 进程"C:\Qt\Qt5.7.1\Tools\mingw530_32\bin\mingw32-make.exe"退出，退出代码 2 。
Error while building/deploying project qgroundcontrol (kit: Android for armeabi-v7a (GCC 4.9, Qt 5.7.1))
When executing step "Copy application data"

gigorvd, 2017-06-20:
I encountered this error when I tried to compile it on Windows 10. On Ubuntu 16.04 compiled without problems.

DonLakeFlyer, 2017-06-21:
There is likely some way to build Android on windows but the fact is that nobody on the dev team does that. So somebody else is going to have to figure it out. I doubt it is some problem with QGC itself.

closed 2017-06-21 by DonLakeFlyer

pmtolk, 2017-10-13:
My current windows build is also failing with a similar but slightly different message

arm-linux-androideabi-g++: error: obj\moc_MockLinkFileServer.objobj\moc_MockLinkMissionItemHandler.obj: No such file or directory

gitterc, 2021-06-11:
gradle need to download some jar packet,see user/administrator/.gradle,maybe download failed ,make sure download success

gitterc, 2021-06-11:
> gradle需要下载一些jar包，查看user/administrator/.gradle，可能下载失败，确保下载成功

Steps in linking debug\libqgroundcontrol

gitterc, 2021-06-11:
> gradle need to download some jar packet,see user/administrator/.gradle,maybe download failed ,make sure download success

Steps in linking debug\libqgroundcontrol