mavlink/qgroundcontrol #5260: Multiple Video Streams

dogmaphobic, 2017-06-07:
Making an issue so it doesn’t get lost:

The current Video Streaming implementation (GStreamer) is hardcoded to one single stream. There is only one instance of `VideoManager`,  which contains just one instance of everything within.

We need to break that apart so we can create one or *more* instances, allowing for multiple, concurrent streams. There are cases where you may have dual cameras (or a camera with dual streams such as thermal imaging cameras). There are also cases where you want to have the ability to play back a recorded stream off a file without having to close the running stream from the vehicle.

@NaterGator? @jaxxzer? :)

jaxxzer, 2017-06-08:
I'm super into this. There's a bunch of folks who want to put cameras all over their ROVs looking at the manipulator, rearview, downward facing.

>  There are also cases where you want to have the ability to play back a recorded stream off a file without having to close the running stream from the vehicle.

This surprises me, do you have a specific case in mind?

dogmaphobic, 2017-06-08:
The next logical step when recording the video stream to local storage is to play it back. If we have only one instance of Gstreamer, we need to close it and open again with a different pipeline (play off a file).

jaxxzer, 2017-06-08:
> The next logical step when recording the video stream to local storage is to play it back.

Sure, it just doesn't seem like QGC's job to me. The OS ships with a media player.

jaxxzer, 2017-06-08:
I guess maybe to make sure you 'got the shot'. But I still don't know how most would rather go about that: within QGC or outside of it (just another app window).

dogmaphobic, 2017-06-08:
> just another app window

Not on mobile. Things don't go too well if you switch apps on mobile while connected to a vehicle.

lbegani, 2017-06-09:
It will be good to have multiple video streams from multiple cameras mounted on the drone. Also, there should be the choice of protocol(UDP, RTSP etc) to the user over which video can be streamed. Playback of recorded video within the app gives a nice user experience. That means we will be supporting streaming+recording+playback simultaneously.

NaterGator, 2017-06-09:
> This surprises me, do you have a specific case in mind?

Search and rescue. _Did we just spot a person? Rewind that 10 seconds..._

patrickelectric, 2018-04-23:
Hi @dogmaphobic, can you update us with the actual status of this issue ?

dogmaphobic, 2018-04-23:
Dual video _capability_ (or multiple for that matter) has been implemented and it is fully functional. What is lacking is UI. There are some GStreamer issues around some corner cases and I'm having a hard time finding GStreamer experts. I've even posted a paid job request on Upwork. No takers.

dogmaphobic, 2019-03-31:
This is all done.

closed 2019-03-31 by dogmaphobic

yuenlm, 2019-05-09:
Are there any plans to support multiple RTSP streams in QGC? I am running v3.5.0 and it appears you can only select one RTSP Stream at a time.

dogmaphobic, 2019-05-10:
The code is already there but there isn't yet any UI to set it up. It can now handle multiple streams but having one at a time. QGC will need UI support for displaying more than one at the same time. Also, this only works for cameras that support the camera API. There is no support for manually configured cameras.

yuenlm, 2019-05-10:
Ah thank you! I was wondering from your earlier comment "This is all done" if the UI had been finished in some other spot. 

Just to clarify my usage, I am adding the RTSP in the following way:

[image]

I am not sure what you mean by manually configured cameras - I bought an off the shelf IP camera and am able to just type in the RTSP URL and it works. Is this different if there are two cameras? 

I have a working developers build of QGC and I am very interested in helping out if you can point me in the right direction. I assume that the basic Settings->General->Video section needs to be updated to allow the user to enter in another camera (and just to double-check- the internals only support "dual" and not "multiple" streaming video correct?)

I did think of a workaround (although doesn't allow simultaneous streaming). It would be handy if in the "Video Stream Widget" (where you can record etc) there was a drop down that the user could select which stream is being displayed (could have the user save pre-populated "favourites" Stream types & addresses) rather than having the user have to manually edit the Video Stream settings every time they want to switch cameras. Thoughts?

For background information my use-case is a vehicle that can move in many different directions (so it would be nice to be able to see which way the vehicle is going, i.e switch between different cameras or show them all at once).

dogmaphobic, 2019-05-10:
> It would be handy if in the "Video Stream Widget" (where you can record etc) there was a drop down that the user could select which stream is being displayed

This is already there:

[image]

But all this requires an "intelligent" camera controller running the [MAVLink camera API](https://mavlink.io/en/services/camera.html). That's how both camera and video are discovered and self configured.

Also note that some of this work only went into the new [qgc4](https://github.com/mavlink/qgroundcontrol/tree/qgc4) branch and not to master yet. That branch is still in a WIP state and once it's all done, it will be merged to master.

There is also initial support for thermal cameras. If a camera identifies itself as having a thermal stream, you have more options allowing the stream to be overlaid over the primary video (with opacity options) as well as full screen and PIP.

There is no support for multiple, manually configured cameras. That is, cameras that do not support the MAVLink camera API.

yuenlm, 2019-05-10:
Ohhhh OK! This really clears things up!

Thank you very much!

jaxxzer, 2019-06-06:
Is there any capability here (or planned) to display multiple video streams simultaneously?

dogmaphobic, 2019-06-06:
> Is there any capability here (or planned) to display multiple video streams simultaneously?

It's already there in code form. That is, you can create any number of the video receiver instances (with its own video source) and have its display shown concurrently with others on the screen. QGC does not have support for it within its UI but only because so far this hasn't been a standard nor common.

jaxxzer, 2019-06-06:
Does the mavlink camera api need to be involved in consideration of this? Or could this be accomplished with a patch in QGC only?

dogmaphobic, 2019-06-06:
Only for self discovery and auto configuration. You can do all this manually. You would create a new instance of `VideoReceiver`, some UI to set its video source (or hardcode it), and finally, a new (QML) instance of `VideoItem` pointing to this new VideoReceiver.

shravannair, 2020-06-19:
@dogmaphobic Hey, so I was planning on adding the UI for the multiple video streaming feature and had some doubts. 
Just to be clear, does QGC have the code for the feature written in it already, or does it just have the capability for us to add the code to it? And if the latter is the case, do I just have to create a new instance in the existing `VideoReceiver.h` file?

dogmaphobic, 2020-06-19:
The code has changed quite a bit since but the idea remains the same. I have not kept track of all the changes. Here is where two video receivers are created within QGC:
https://github.com/mavlink/qgroundcontrol/blob/8512a64170dafe9f8dfd49762cd9dbf95064ed00/src/VideoManager/VideoManager.cc#L115-L116

The second receiver is used for "thermal" cameras (they usually have two streams) and it is hardcoded as such. Meaning, if the camera announces itself as a thermal camera with two streams (and the thermal video is flagged as such), the second video receiver is shown automatically. If you want to add your own second (third/fourth/etc.) video stream, you can follow the same logic. You will need to create your own video display window for each of them though.

patrickelectric, 2020-06-19:
@shravannair Check https://github.com/mavlink/qgroundcontrol/pull/8017
I saw it running and managing more than 32 pipelines dynamically

shravannair, 2020-06-19:
Aah, okay. I'll look into them. Thanks a lot! @dogmaphobic @patrickelectric

Prasad28-97, 2021-08-20:
> This is already there:
>
> [image]
>
> But all this requires an "intelligent" camera controller running the [MAVLink camera API](https://mavlink.io/en/services/camera.html). That's how both camera and [...]

Please specify the version of Qgroundcontrol where this multiple camera streaming feature is implemented.The link you have dropped (qgc4) isnt working.

socalengineer, 2022-03-09:
Hi,

The use case I have for this is to have multiple quality video streams for a drone, 720p, 1080p, etc. And based on the data link switch between the different quality streams. Is it possible to switch with a hotkey or etc when I deem the use of the better video stream?

Thanks!