mavlink/qgroundcontrol #5261: Enable More Joystick Channels

SwiftGust, 2017-06-08:
Hello, I'm Using QGC for my joystick enabler in SITL.
But I'm having trouble with joystick for only having 4 channels
such as flight mode change or more channels
even mine has lots of buttons and sliders.

I would appreciate if there's more available channels.

Thanks.

jaxxzer, 2017-06-08:
You can use the buttons for things like flight mode change, but your autopilot firmware must support it (MANUAL_CONTROL mavlink message).

SwiftGust, 2017-06-09:
That's good to hear, but what I think required is literally more channels so, able to manipulate either manually or automatically more servos or gimbaled cameras in the 3D simulations like Gazebo

AlexandreBorowczyk, 2017-07-05:
I'm also interested to know if the xbox buttons can be used as rc channel. I'm running the px4 in sitl with gazebo.

hamishwillee, 2017-07-05:
I've just updated the [Joysticks](https://docs.qgroundcontrol.com/en/SetupView/Joystick.html) doc in the user guide (based on experimentation, not insider knowledge). On the PS4 controller I see HEAPS of buttons I can map, and mapping them works really well. On Taranis I see only 8 buttons I can map (BUT, when I toggle my taranis switches nothing happens). 

Upshot, I "guess" that QGC probably supports all the channels that it sees in the connected joystick. However some joysticks will expose more of these buttons than others OR the protocol for sharing this information with the OS/QGC isn't consistent so you will get different levels of support.

@DonLakeFlyer would be good to have clarification on this.

@AlexandreBorowczyk I think the easiest way to find out is to give it a try. If it works, let me know what Xbox controller you're using and we can add it to the docs.

DonLakeFlyer, 2017-07-06:
QGC has little control over what buttons get exposed or not. We use the SDL library for joystick support. So we get whatever it supports.

hamishwillee, 2017-07-10:
Closing this as it is outside QGC control. I added a note to the top of the [appropriate page](https://docs.qgroundcontrol.com/en/SetupView/Joystick.html) in the  user guide to make this clear

closed 2017-07-10 by hamishwillee