mavlink/qgroundcontrol #5263: Sensor Cal: Change "Sensor Settings" to "Set Orientations"

DonLakeFlyer, 2017-06-08:
PX4 uses Set Orientations. ArduPilot uses Sensor Settings. Should be the same thing,.

hamishwillee, 2017-06-08:
Yes.
The only question here is what to call the menu text.

- Ardupilot allows disabling an external compass in the settings too. It might be worth moving that option into compass calibration and simply not displaying external compass in orientation options on ardupilot if the compass is disabled. Anyway, this is a non-orientation setting
- This covers both Autopilot and external compass. On one hand a label that reflects autopilot will be more intuitive to users, but if we use that then people who want to set the compass orientation wouldn't know where to look.

I tend to like "Sensor Orientations"

DonLakeFlyer, 2017-06-20:
I looked at this. I think Sensor Orientation's is better. But it doesn't reflect what is in there for ArduPIlot since it is more than orientations. So I'm going to leave this alone for now.

closed 2017-06-20 by DonLakeFlyer